mavlink/qgroundcontrol #9128: Connecting Raspberry pi to QGroundControl

sayansarkar2207, 2020-10-20:
I want to use Raspberry pi as a vehicle controller and control it through QGroundControl.

This project will help us to use raspberry pi directly as the vehicle controller.
Currently i am using the UDP ports on the raspberry pi to connect it with QGroundControl.

Issues I am facing
I successfuly used the UDP ports of the raspberry pi to connect it with my laptop and successfully connected. But when i try to connect the raspberry pi with QGroundControl it does not respond or connect with the Pi.

I have also tried it many times by using different port numbers.

I am making my Raspberry Pi as the server and the QGround as the host.

I had also written a python code for the Raspberry pi to connect it as a server. Here I am giving my code and i am also giving it as an [attachment](https://drive.google.com/file/d/15qLicntPZSC6QcYFM22YYcMWPFmX7HM_/view?usp=sharing)

Anyone can suggest me a way how to connect it, through UDP or through any other method.
Anything you can suggest me, please fell free to reply.

Program Code:

import socket

host = ''
port = 4560

storedValue = "Hi, how are you"

def setupServer():
    s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
    print("Socket created.")
    try:
        s.bind((host, port))
    except socket.error as msg:
        print(msg)
    print("Socket bind comlete.")
    return s

def setupConnection():
    s.listen(1) # Allows one connection at a time.
    conn, address = s.accept()
    print("Connected to: " + address[0] + ":" + str(address[1]))
    return conn

def GET():
    reply = storedValue
    return reply

def REPEAT(dataMessage):
    reply = dataMessage[1]
    return reply

def dataTransfer(conn):
    # A big loop that sends/receives data until told not to.
    while True:
        # Receive the data
        data = conn.recv(1024) # receive the data
        data = data.decode('utf-8')
        # Split the data such that you separate the command
        # from the rest of the data.
        dataMessage = data.split(' ', 1)
        command = dataMessage[0]
        if command == 'GET':
            reply = GET()
        elif command == 'REPEAT':
            reply = REPEAT(dataMessage)
        elif command == 'EXIT':
            print("Our client has left us :(")
            break
        elif command == 'KILL':
            print("Our server is shutting down.")
            s.close()
            break
        else:
            reply = 'Unknown Command'
        # Send the reply back to the client
        conn.sendall(str.encode(reply))
        print("Data has been sent!")
    conn.close()

s = setupServer()

while True:
    try:
        conn = setupConnection()
        dataTransfer(conn)
    except:
        break

DonLakeFlyer, 2020-10-20:
GitHub issues are for bugs. Use either forums or slack for general questions.

closed 2020-10-20 by DonLakeFlyer

sayansarkar2207, 2020-10-20:
Can you please give me the link of the forum

dogmaphobic, 2020-10-21:
> Can you please give me the link of the forum

See the "Support" section at the bottom of the main page:
http://qgroundcontrol.com