mavlink/qgroundcontrol #9138: Parameters are missing from firmware. You may be running a version of firmware QGC does not work correctly with your firmware or your firmware has a bug in it.

abhiramdapke, 2020-10-23:
## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 18.04
- QGC Version: Latest Version recommended by Dronecode
- QGC build: Stable on command-line
- Flight Controller: Pixhawk 2.1 with the Black cube
- Autopilot (with version): PX4 firmware 4.0.3 CubeBlack

## Log Files and Screenshots

[image]

## Detailed Description
I am trying to calibrate the Pixhawk 2.1 with the Black cube flight controller. I am getting an error that is the name of this topic. I tried installing the latest QgroundControl and install the Pixhawk v1.11 firmware again but still, the error is there.

When I try to calibrate the radio using a joystick, It doesn't detect the joystick and gives the same error.
Also, while calibrating the compass, previously I was getting all the different configuration blocks but now I am not getting any. I am thinking there is an issue with the firmware. Let me know if that could be a problem.

Further, is it possible that the black cube has a specific firmware installed and parameters tuned and if I upgrade the firmware, the parameters cannot be tuned back?

Also, I have some doubts regarding the same.
1. Can I fly a real drone like  DJI but one that uses PX4 autopilot (i.e. open-source software) without the RC transmitter or joystick?
2. Can I calibrate all the parameters and simulate the drone (in HITL mode)on the computer using Qgroundcontrol without the GPS calibration?
3. I have a pixhawk 2.1 carrier board with a black cube, what are the parameters that could be changed to fit the requirements of a real drone. Let's say I have a black cube with certain parameters and a 4.0.3 firmware. Now, I flashed it with different firmware. Is it possible to get the parameters back as I got the firmware back? I believe that would depend on which parameters have been changed. Let me know if I am right.
4. Can I fly multiple drones with ROS Gazebo and PIxhawk Black cube each having a different planned path?

DonLakeFlyer, 2020-10-23:
* PX4 Firmware recently added support for multiple batteries. In doing so the names for the battery parameters changed to look like this BAT#_* where # is replaced with 1, 2 and so forth.
* You are running QGC Stable which was released prior to these firmware changes. Hence QGC Stable knows nothing about these new param names.
* The firmware was supposed to have implemented back compat support on the param names which should have kept QGC Stable working. I don't think 1.11 has that in it. 1.12 did but originally that support was buggy. I believe they just released a new stable point release a few days back which has the fixes for it. It's 1.12.something. So updated to latest firmware stable should make QGC work fine.

So anyway, the fix is to update to latest firmware stable since the bug is there not in QGC.

As far as your other questions:

> 1. Can I fly a real drone like  DJI but one that uses PX4 autopilot (i.e. open-source software) without the RC transmitter or joystick?

Yes.

> 2\. Can I calibrate all the parameters and simulate the drone (in HITL mode)on the computer using Qgroundcontrol without the GPS calibration?

HITL is no longer supported. Only SITL.

> 3\. I have a pixhawk 2.1 carrier board with a black cube, what are the parameters that could be changed to fit the requirements of a real drone. Let's say I have a black cube with certain parameters and a 4.0.3 firmware.

You seem to be mixing PX4 and Ardupilot firmware in your questions. 4.0* numbering is ArduPIlot firmware not PX4. You can always save your parameters to a file and reload them whenever you want. Parameters differ between firmwares.

> 4\. Can I fly multiple drones with ROS Gazebo and PIxhawk Black cube each having a different planned path?

No HITL with QGC. But QGC does support multiple vehicles. So if you can get a SITL setup to do that then QGC will work fine with it.

closed 2020-10-23 by DonLakeFlyer

abhiramdapke, 2020-10-27:
Thanks a lot for your explanation, DonLakeFlyer.
I had a couple more questions regarding this.

How is it possible to fly a real drone without using PX4 autopilot with a pixhawk carrier board and a cube controller without the RC transmitter or joystick because whenever I try to start a mission, Qgroundcontrol always has those red signals that state the radio, power, etc not calibrated?
Also, is it possible to automate the drone by Qgroundocntrol or I need a Tx2, Rasberry Pi to run the Path Planning scripts?

One more question is that if I have all the parameters of a particular drone on either Mission Planner/Qgroundcontrol, can I run a simulation of it in Gazebo or any other simulator? I am asking this because for the Ardupilot firmware, there is no airframe that resembles closely the drone that I have(in SITL) and I want to perform a simulation of the actual drone on the computer. So, do I need a Flight Dynamic Model of the actual drone(i.e. a .stl,.obj, Collada format file) to simulate it in Gazebo or any other simulator?

Also, in the last answer, you said if you get the SITL setup, then Qgc will work fine. I have the SITL setup and I performed some simulations on it in Gazebo as well but they are not of the actual drone. They are of generic quadcopters or any other generic airframes. 

Thank you.

DonLakeFlyer, 2020-10-27:
You should at those questions on a firmware forum. QGC isn't in control of that. The PX4 dev guide (maybe user guide don't remember) guide has instructions on how to set things up without an RC transmitter. You should look through the PX4 user/dev guide to understand how things work. And then if you still have questions the place would be a PX4 Forum or the PX4 slack channel.

abhiramdapke, 2020-10-29:
Thanks a lot, Don. I really appreciate it.

Igor-Misic, 2021-01-07:
@DonLakeFlyer does QGC really need to expect BAT_V_CHARGED, BAT_N_CELLS, BAT_V_EMTY, BAT_V_LOADDROP  when BAT1_SOURCE = external?  

I am trying to figure out when we will be ready to use an external battery source. At the moment I am at FW v1.11.3 and QGC  v4.0.11

[image]

DonLakeFlyer, 2021-01-07:
Use a daily build instead. This has been fixed. New QGC stable is about a week away. At this point daily = stable.

Igor-Misic, 2021-01-07:
Same error with daily when BAT1_SOURCE = external. 

[image]

DonLakeFlyer, 2021-01-07:
> BAT1_SOURCE = external

Hmm, ok that wasn't what the fix was for. That is a PX4 feature I haven't seen yet. Can you create a new issue for this and then I can get firmware folks involved to figure out how this is supposed to work.

Igor-Misic, 2021-01-08:
New issue created:   #9312