mavlink/qgroundcontrol #9140: Error message : when I build the project

rien3271, 2020-10-27:
## Steps to Reproduce

- I have downloaded the software QGround control and QT as explaining in the [developer guide](https://dev.qgroundcontrol.com/master/en/getting_started/)

- I have then install some dev package that could solve the issue like `apt-get install qt5-qtlocation-devel` but I get the same error

## Current Behavior
When I  build qgroundcontrol project with QT Creator I get the following error : 

`:-1: error: Unknown module(s) in QT: location quickcontrols2 texttospeech multimedia location-private positioning-private charts x11extras waylandclient`

## System Information
- Operating System: Ubuntu 20.04.1
- QGC Version: 4.0
- QT Version : qt5

Thank you in advance

patrickelectric, 2020-11-20:
Something is wrong with your installation or configuration.

closed 2020-11-20 by DonLakeFlyer

mfry90, 2022-02-22:
@rien3271 Did you find a solution to your issue? I'm running into the same problem after downloading QT from Git, building and installing from source, and then downloading QGroundControl from Git and trying to build.