mavlink/qgroundcontrol #9141: Settings not saved on MacOS

tobiasbischoff, 2020-10-27:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
Im changing Speed under General from Meters/second to Kilometers/hour - exit the App and restart and its changed back to Meters/second. Changes there seem not to get saved, there is also no save button.

## Current Behavior
How can i make Setting changes survive an app restart?

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1.
2.
3.
4.

## System Information
When posting bug reports, include the following information
- Operating System: Mac OS 10.15.7
- QGC Version: 4.0.10
- QGC build: release
- Flight Controller: [e.g., Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.]
- Autopilot (with version): [e.g., PX4 1.8.1, Ardupilot Copter 3.6.1]

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

DonLakeFlyer, 2020-10-29:
This doesn't reproduce for me. Is it only that setting that doesn't stick?

tobiasbischoff, 2020-10-29:
> This doesn't reproduce for me. Is it only that setting that doesn't stick?

No, all Settings don't stick, even Window Position, Map Typ etc. Very strange, my Desktop Mac does not have that issue, and its same OS version, roughly the same apps installed. Only difference between both systems is that the MacBook that has the Issue has full disk encryption enabled.

DonLakeFlyer, 2020-10-29:
If you look at the QGC Application Settings/Console output. It should tell you where your settings are being stored. Also look in there for any error messages which may indicate a failed file write or something like that. Maybe the location where it is trying to store settings is write protected or something?

First line in console:
[image]

tobiasbischoff, 2020-10-29:
Yes, there is definatly some Settings related Error, full Log in attachement

[E] at file:///Applications/QGroundControl.app/Contents/Resources/qml/QtQuick/Dialogs/DefaultFileDialog.qml:102 - "file:///Applications/QGroundControl.app/Contents/Resources/qml/QtQuick/Dialogs/DefaultFileDialog.qml:102:33: QML Settings: Failed to initialize QSettings instance. Status code is: 1"

[123.txt](https://github.com/mavlink/qgroundcontrol/files/5461339/123.txt)

DonLakeFlyer, 2020-10-29:
It's some sort of permissions problem on that machine. You can see it here:
```
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/QGCApplication.cc:267 - "Settings location "/Users/tobi/.config/org.qgroundcontrol/QGroundControl.ini" Is writable?: false"
```

That location should be writable.

tobiasbischoff, 2020-10-29:
Oh, that directory was totally broken for some reason. chown - r and chmod -r 755 did the trick. Works now, thanks alot!

DonLakeFlyer, 2020-10-29:
Re-opening since the fact that QGC checked for settings location to be writable and found it wasn't and then didn't tell the user anything about that doesn't seem like a good thing.

booo, 2022-09-15:
There is debug output such as:

```
Settings location "/home/borgers/.config/QGroundControl.org/QGroundControl Daily.ini" Is writable?: false
```

Do we need a popup or something? We could add a patch similar to this:

```diff
diff --git a/src/QGCApplication.cc b/src/QGCApplication.cc
index 74e37c2bf..474b95d54 100644
--- a/src/QGCApplication.cc
+++ b/src/QGCApplication.cc
@@ -280,6 +280,10 @@ QGCApplication::QGCApplication(int &argc, char* argv[], bool unitTesting)
     QSettings settings;
     qDebug() << "Settings location" << settings.fileName() << "Is writable?:" << settings.isWritable();
 
+    if (!settings.isWritable()) {
+      _exitWithError(QString("Settiong location '" + settings.fileName() + "' is not writable."));
+    }
+
 #ifdef UNITTEST_BUILD
     if (!settings.isWritable()) {
         qWarning() << "Setings location is not writable";
```