mavlink/qgroundcontrol #9149: PPA: Failed to build QGC

thopiekar, 2020-10-28:
As you may know, I'm building Debian packages in my "drone" PPA.
Builds are failing for some time already with this:

```
compiling /<<PKGBUILDDIR>>/src/GPS/Drivers/src/ubx.cpp
compiling /<<PKGBUILDDIR>>/src/GPS/Drivers/src/sbf.cpp
/<<PKGBUILDDIR>>/src/GPS/Drivers/src/ashtech.cpp: In member function ‘int GPSDriverAshtech::handleMessage(int)’:
/<<PKGBUILDDIR>>/src/GPS/Drivers/src/ashtech.cpp:589:28: error: ‘struct satellite_info_s’ has no member named ‘prn’
  589 |    memset(_satellite_info->prn,       0, sizeof(_satellite_info->prn));
      |                            ^~~
/<<PKGBUILDDIR>>/src/GPS/Drivers/src/ashtech.cpp:589:66: error: ‘struct satellite_info_s’ has no member named ‘prn’
  589 |    memset(_satellite_info->prn,       0, sizeof(_satellite_info->prn));
      |                                                                  ^~~
/<<PKGBUILDDIR>>/src/GPS/Drivers/src/ashtech.cpp:620:22: error: ‘struct satellite_info_s’ has no member named ‘prn’
  620 |     _satellite_info->prn[y + (this_msg_num - 1) * 4]       = sat[y].prn;
      |                      ^~~
make[1]: *** [Makefile:22040: obj/ashtech.o] Error 1
make[1]: *** Waiting for unfinished jobs....
/<<PKGBUILDDIR>>/src/GPS/Drivers/src/ubx.cpp: In member function ‘int GPSDriverUBX::payloadRxAddNavSat(uint8_t)’:
/<<PKGBUILDDIR>>/src/GPS/Drivers/src/ubx.cpp:1291:22: error: ‘struct satellite_info_s’ has no member named ‘prn’
 1291 |     _satellite_info->prn[sat_index]    = static_cast<uint8_t>(_buf.payload_rx_nav_sat_part2.prn);
      |                      ^~~
/<<PKGBUILDDIR>>/src/GPS/Drivers/src/ubx.cpp: In member function ‘int GPSDriverUBX::payloadRxAddNavSvinfo(uint8_t)’:
/<<PKGBUILDDIR>>/src/GPS/Drivers/src/ubx.cpp:1350:22: error: ‘struct satellite_info_s’ has no member named ‘prn’
 1350 |     _satellite_info->prn[sat_index]       = static_cast<uint8_t>(_buf.payload_rx_nav_svinfo_part2.prn);
      |                      ^~~
/<<PKGBUILDDIR>>/src/GPS/Drivers/src/ubx.cpp: In member function ‘int GPSDriverUBX::payloadRxDone()’:
/<<PKGBUILDDIR>>/src/GPS/Drivers/src/ubx.cpp:1733:16: error: ‘M_PI_F’ was not declared in this scope
 1733 |     heading *= M_PI_F / 180.0f; // deg to rad, now in range [0, 2pi]
      |                ^~~~~~
make[1]: *** [Makefile:22049: obj/ubx.o] Error 1
make[1]: Leaving directory '/<<PKGBUILDDIR>>/build'
dh_auto_build: error: cd build && make -j4 returned exit code 2
```

Full log:
https://launchpadlibrarian.net/503980652/buildlog_ubuntu-focal-amd64.qgroundcontrol_4.0.10-0-0drone0ppa0~202010281007~5968~packaging5956~ubuntu20.04.1_BUILDING.txt.gz

thopiekar, 2020-10-28:
Ok, not really related to this project, but the GPS drivers by the PX4 people.
Must be caused by one of the two:
https://github.com/PX4/PX4-GPSDrivers/commit/e33d72bbcded623c3da718d8ce098ea9779f5921
https://github.com/PX4/PX4-GPSDrivers/commit/d190e96a9e7ab6f58eb072032c0b14fbb8821ba5

In this matter pinging: @bkueng

dogmaphobic, 2020-10-28:
Dumb question: have you updated the GPS submodule? Or checked out a newer commit than the one used by QGC? The error above on `ashtech.cpp:589:28` doesn't match the existing code (the submodule commit used by QGC).

dogmaphobic, 2020-10-28:
https://github.com/PX4/PX4-GPSDrivers/blob/318562bac7cd55e6cfafd4694094343a70e4f322/src/ashtech.cpp#L589

bkueng, 2020-10-29:
Yes you're using an incorrect submodule commit.

DonLakeFlyer, 2020-10-29:
@bkueng What is the expectation with respect to when the gps submodule in QGC gets updated and by whom? I personally never touch it. Should I assume firmware folks (you?) will update as needed?

Closing this since it appears it's using the wrong gps code.

closed 2020-10-29 by DonLakeFlyer

bkueng, 2020-10-30:
Yes I generally do that, but didn't have a reason to do that lately.

DonLakeFlyer, 2020-10-30:
Great, thanks @bkueng