mavlink/qgroundcontrol #9162: Missing QML widgets option in last releases

dell640, 2020-11-05:
We are using QML widgets for camera control via MavLink (via AIR Commander Entire).
From all it seems this feature has been removed since v4, which is bad news.

Plugin from airpixel.cz is able to fully control almost any camera mounted on the drone via MvLink (all exposure parameters) which can not be replaced by any other QGC feature.

If possible, please add this option back to QGC menu.

[image]

closed 2020-11-05 by DonLakeFlyer

dell640, 2020-11-05:
Thanks for fast response! Can you share some tutorial / examples of changing QGC GUI without reprogram of source code and compilling own QGC?
("do things like override the existing camera control resources if you like and put your own thing in there")

DonLakeFlyer, 2020-11-05:
> Can you share some tutorial / examples of changing QGC GUI without reprogram of source code and compilling own QGC?

That is no longer supported. You need to build your own version using the custom build architecture which is talked about in the qgv dev guide.

The new mavlink camera protocol is the mechanism for cameras to expose UI for settings: https://mavlink.io/en/services/camera.html#:~:text=Message%2FEnum%20Summary%20%20%20%20Message%20%20,camera%20mode%20information.%20%2018%20more%20rows%20. That is already fully supported by QGC.