mavlink/qgroundcontrol #9164: Saved survey won't load without .plan ext - Linux

mj-saunders, 2020-11-06:
## Expected Behavior
Opening a saved survey mission should load all relevant settings (regardless of file extension?)

## Current Behavior
As far as I can tell, no settings are being loaded unless `.plan` extension is present

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Create new Survey mission
2. Save file (linux app does not automatically add `.plan`)
3. JSON in file looks good
4. Restart app and open saved file
5. Nothing gets loaded

## System Information
When posting bug reports, include the following information
- Arch Linux `5.9.2-arch1-1`
- QGC Version: `4.0.11`
- QGC build: built from source via https://aur.archlinux.org/packages/qgroundcontrol/
- Flight Controller: n/a
- Autopilot (with version): n/a

## Detailed Description
New user to the app. Testing survey flight planning. Can create and save a mission.
Opening the saved file in text editor shows what looks to be reasonable JSON.
Attempted to open previously saved mission, but no settings appear to be updated.
While writing this, discovered the issue to be a lack of `.plan` extension.

## Log Files and Screenshots
- As far as I can tell, the lack of extension leads to app silently not attempting to load the file

DonLakeFlyer, 2020-11-06:
Thanks so much for the detailed bug report. The problem where you could save a plan file without an extension had already been fixed previously. But there was still more to it. All of these fixes are in Daily builds.

closed 2020-11-06 by DonLakeFlyer

mj-saunders, 2020-11-06:
@DonLakeFlyer No problem. Thank you. Sorry I didn't manage to find the previous issue(s).
I had some trouble building from latest git source - but will try again when I have the time.