mavlink/qgroundcontrol #9177: VideoReceiver::videoSizeChanged - signal emitted but with bad size value

tzuriel-spearuav, 2020-11-15:
## Expected Behavior
The videoSizeChanged signal should be emitted with width and height information of the video (resolution of frames)

## Current Behavior
The signal is emitted but contains wrong information.
Seems like fields "width" and "height" are not present in the capabilities of the pad queried. As a result VideoManager::videoSize contains wrong information

## Steps to Reproduce:
1. Load QGC with SITlL-gazebo plugin
2. Configure video input for UDP H264
3. Set breakpoint at lambda for VideoReceiver::videoSizeChanged (https://github.com/mavlink/qgroundcontrol/blob/4bd3e83efa7652bbea8fbd3dd89011886b39bd27/src/VideoManager/VideoManager.cc#L163-L166)
4. Inspect retrieved size - values are incorrect

## System Information
When posting bug reports, include the following information
- Operating System: Windows10
- QGC Version: Daily 14-Nov-2020
- QGC build: daily
- Flight Controller: SITL

DonLakeFlyer, 2020-11-23:
I looked and there are no current consumers of videoSize/videoSizeChanged property/signal. So not super surprised it doesn't work. Not sure why it was added in the first place since it is never used. Happy to take a pull that fixes it. In general I tend to remove unused code since it just sits around and breaks.

tzuriel-spearuav, 2020-11-25:
Not sure I fully understand gstreamer caps and the negotiation process, but seems to me that GstVideoReceiver::_addVideoSink() is querying an un-negotiated element (src pad). Instead if I'm changing the code to query _decoderValve it seems to get the correct width/height.
I suggest changing the following line https://github.com/mavlink/qgroundcontrol/blob/3b7dd6e87c9600d184e9bfe59cf53500d5639a12/src/VideoReceiver/GstVideoReceiver.cc#L1077
to 
`GstPad* valveSrcPad = gst_element_get_static_pad(_decoderValve, "src");`
`GstCaps* caps = gst_pad_query_caps(valveSrcPad, nullptr);`

DonLakeFlyer, 2020-11-27:
@patrickelectric ?

patrickelectric, 2020-11-28:
I can take a look monday, but the suggestion looks good.

patrickelectric, 2020-12-01:
@tzuriel-spearuav I was able to replicate and your suggestion fixed the issue, I have created a PR with the proposal.

PR #9231 (2020-12-02): GstVideoReceiver: Fix video size signal
Extracting video size from decoderValve is more guaranteed.

Fix #9177
closed 2020-12-02 by DonLakeFlyer