mavlink/qgroundcontrol #9185: Bring back dumb camera trigger support

DonLakeFlyer, 2020-11-20:
With the changes to the new camera widget the ability to trigger a dumb camera using `MAV_CMD_DO_DIGICAM_CONTROL` was lost. Need to figure out how to get that ability back. It's a little tricky from a user model standpoint since video streaming has camera trigger to capture a frame from the video feed as well even for camera which are not using the new mavlink camera protocol.

moreba1, 2020-11-22:
Do you can also add a button for start/stop camera triggering?
 it can be used during mission and for tests on the ground.

DonLakeFlyer, 2020-11-26:
> Do you can also add a button for start/stop camera triggering?
> it can be used during mission and for tests on the ground.

I'd rather keep it simple. If you want something fancy you can fly a mission

closed 2020-11-26 by DonLakeFlyer