mavlink/qgroundcontrol #9193: Android 10 + QGC 4.0.11.64: No serial ports available

hagfelsh, 2020-11-21:
I'm having the exact same issue as described in detail in #4055 

I'm opening this bug anew because 4 years have passed since the creation of its predecessor.

I've tried with two entirely different radios: a 3DR 2.4ghz and some sort of unnamed 433mHz unit.

The phone is a Samsung Galaxy S10 note+

Problem statement:
Upon connection of either radios mentioned above, android asks if I'd like to open QGroundControl for use with this device.  Hitting OK opens QGC.  I add a connection in QGC and scroll down to find the serial port chooser indicates "No serial ports available".

#  Expected Behavior
I'd expect to see the 1 serial port I connected via USB listed

## Current Behavior
"No serial ports available" is presented instead

## Steps to Reproduce:
1. Plug in either of the radios mentioned above
2. Create a new serial connection
3. Find 'no serial ports available' instead of a device option.

## System Information
- Operating System: Android 10 build QP1A.190711.020.N975USQS5DTJ8, kernel 4.14.117-19541081
- QGC Version: 4.0.11.64
- QGC build: Acquired from Google Play
- Flight Controller: Pixhawk
- Autopilot (with version): Plane 4.0.6

## Log Files and Screenshots
- USB device dump screenshots attached
- QGC AndroidSerialPortLog, LinkManagerVerboseLog, SerialLinkLog, and QGCSerialPortInfoLog attached
[Qgclogs.txt](https://github.com/mavlink/qgroundcontrol/files/5578266/Qgclogs.txt)
[image]
[image]

DonLakeFlyer, 2020-11-21:
> I add a connection in QGC and scroll down to find the serial port chooser indicates "No serial ports available".

There should be no need to add a manual comm link. Just plugging in the radio should be all you need to do. Anything past that should be automatic. If a vehicle shows up on the radio it should just show up in QGC. Is that not working?

DonLakeFlyer, 2020-11-21:
Can you try the daily build instead? There is a bunch of serial port changes in there. Just plug in the radio, let QGC boot from that and then do nothing else in QGC. Don't try to create a comm link yourself. Then turn your vehicle on on the other side of the radio. Capture the same logs for that sequence.

hagfelsh, 2020-11-28:
Somehow this particular sequence of steps must have eluded me.  I feel like I've tried every permutation but clearly I missed one!

@DonLakeFlyer just as you said: Just plugging it in should be all that's necessary.  Indeed it was, after a fashion!

1. Energize aircraft & radio.  Give the pix a moment to begin sending mavlink frames
2. Start QGC on the tablet
3. Connect the radio...
4. Magic

Seriously--I can't imagine that I've not followed these steps in the past, but the evidence speaks for itself.  Thank you for humoring a silly not-problem!

closed 2020-11-28 by hagfelsh