mavlink/qgroundcontrol #9200: feature request : land command at specied location in guided mode

moreba1, 2020-11-22:
This feature can be used in an emergency condition that we will land drone immediately. currently, we can use land mode in the current location but maybe the attitude and current location were not good. so we must be able to immediately select a safe land location on the map and command drone fly toward it.
for VTOLs waypoint must be VTOL land.

DonLakeFlyer, 2020-11-23:
So fly to the location you want using the method of you choice and Land. Why does there need to be anything other than that?

moreba1, 2020-11-23:
> So fly to the location you want using the method of you choice and Land. Why does there need to be anything other than that?

because when plane reach near that waipoint start loiter around it and have roll angle.

closed 2026-10-04 by github-actions[bot]