mavlink/qgroundcontrol #9202: QGC window opening at saved screen coordinates even if they become invalid

vincentdavoust, 2020-11-23:
## Expected Behavior
QGC window always opens on visible screen even when displays are changed / removed

## Current Behavior
QGC is launched properly but is not displayed. (clicking on the icon loses focus on other windows but does not display, using windows-tab shows window miniature but cannot switch it to visible on screen)

## Steps to Reproduce:
1. Have dual display
2. Open QGC window and position it on screen 2 (the display on the right, could be screen 1 if main screen is switched)
3. Close QGC
4. Disconnect screen 2
5. Relaunch QGC

## System Information
When posting bug reports, include the following information
- Operating System: Windows 10
- QGC Version: 4.0.11
- QGC build: stable

## Detailed Description
I believe the issue comes from using the saved window coordinates in `%AppData%/QGroundControl/QGroundControl.ini [MainWindowState]` without checking that they are within 
`0 <= x < QApplication::desktop()->screenGeometry().width() - width`
and
`0 <= y < QApplication::desktop()->screenGeometry().height() - height`
and also probably constraining width and height to the full display size as a maximum.

vincentdavoust, 2020-11-23:
I am planning to make a fix/PR for this shortly, except if this is already something that is being handled ?
Also any input on where to handle this the cleanest would be appreciated.

DonLakeFlyer, 2020-11-23:
This ends up being a Qt thing. I think folks have tried fixing it in the past and failed. I believe because there was some Qt support missing to make it work. That was a while back though. It's possible Qt 5.12 has what is needed to make this work.

vincentdavoust, 2020-11-25:
I think I've fixed it with an `if` in MainWindowSavedState.qml, making th PR now.

closed 2026-10-06 by github-actions[bot]