mavlink/qgroundcontrol #9208: 'mavlink_types.h' file not found

devd77, 2020-11-24:
--I have cloned this repository on my system by "git clone https://github.com/mavlink/qgroundcontrol.git --recursive"
--and then run command: git submodule update
--then it on QT Creator
--Build for iOS Simulator Debug
--Then i have found "mavlink_types.h' file not found" this issue.

It was working before, but in new clone i have found this issue
Anyone can tell me the solution
Thanks in Advance :)

please see the attachment image
[image]

patrickelectric, 2020-11-24:
Try `git submodule update --init --recursive`

devd77, 2020-11-24:
i have tried, not working...same problem

DonLakeFlyer, 2020-11-24:
As stated in the issue template. This is not the place for build problems.

closed 2020-11-24 by DonLakeFlyer