mavlink/qgroundcontrol #9209: Ardupilot: Vehicle Fact data goes stale after reconnect

dakejahl, 2020-11-24:
When using Ardupilot and QGC, after a disconnect/reconnect event (changing batteries on the vehicle for example) the Vehicle Fact data never updates to new values, it just shows the old values.

DonLakeFlyer, 2020-11-29:
This is most likely caused by the fact that some vehicles types (don't remember which ones) don't automatically start telemetry streams on vehicle boot. The GCS has to request them. That has never made much sense to me. And you can see how it complicates cases like this. QGC only requests streams to start (and ArduPilot only thing) on a vehicle connect. Haven't to create some system which detects telemetry things suddenly disappearing and maybe having to try to ask the vehicle to pretty-please send the telemetry again seems like insanity to me.

dakejahl, 2020-11-29:
@auturgy are you able to shed some insight on this?

DonLakeFlyer, 2020-11-30:
@dakejahl It is what it is. It's been like this forever and ground stations just need to write code to deal with it. My frustration is just showing through with respect to having to write code which accomplishes nothing. Instead of spending timer writing code which makes things better.

closed 2021-07-01 by DonLakeFlyer