mavlink/qgroundcontrol #9210: _sendGCSHeartbeat() race condition crash

dakejahl, 2020-11-24:
I believe there is a race condition between 
https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/MultiVehicleManager.cc#L49
and
https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/MultiVehicleManager.cc#L133
Which causes random crashes.

I will look into it more in a few days.

DonLakeFlyer, 2020-12-07:
Google Play upload is working again so we should be able to get crash data from the store soon.

DonLakeFlyer, 2020-12-10:
So far I'm no longer seeing this crash on Google Play. But it needs a bit more time I think to make sure.

DonLakeFlyer, 2020-12-16:
I believe we are good here now:
[image]

closed 2020-12-16 by DonLakeFlyer