mavlink/qgroundcontrol #9211: Google Play daily upload is broken yet again!

DonLakeFlyer, 2020-11-24:
Last update was Nov 8th.

DonLakeFlyer, 2020-11-28:
This is working. There just seems to be a massive delay on approvals on the Google side.

DonLakeFlyer, 2020-12-03:
So Google Play says the upload day for daily is Nov 28th but the build up there is the build from the 8th! No idea what is going on. Reviews on play store are super slow so not sure if QGC problem or play store delay problem.

pompecukor, 2020-12-04:
Any Chance to download from somewhere directly (just the APK) till this is fixed?

DonLakeFlyer, 2020-12-06:
travis output:
```
Uploading package org.mavlink.qgroundcontrolbeta to track production
1433Traceback (most recent call last):
1434  File "./deploy/GooglePlayUpload.py", line 91, in <module>
1435    main()
1436  File "./deploy/GooglePlayUpload.py", line 60, in main
1437    media_body='QGroundControl32.apk').execute()
1438  File "/home/travis/.local/lib/python2.7/site-packages/googleapiclient/_helpers.py", line 134, in positional_wrapper
1439    return wrapped(*args, **kwargs)
1440  File "/home/travis/.local/lib/python2.7/site-packages/googleapiclient/http.py", line 915, in execute
1441    raise HttpError(resp, content, uri=self.uri)
1442googleapiclient.errors.HttpError: <HttpError 403 when requesting https://androidpublisher.googleapis.com/upload/androidpublisher/v3/applications/org.mavlink.qgroundcontrolbeta/edits/18310404915872232656/apks?uploadType=media&alt=json returned "Your app currently targets API level 28 and must target at least API level 29.". Details: "Your app currently targets API level 28 and must target at least API level 29.">
```

DonLakeFlyer, 2020-12-06:
> Any Chance to download from somewhere directly (just the APK) till this is fixed?

https://qgroundcontrol.s3-us-west-2.amazonaws.com/master/QGroundControl32.apk
https://qgroundcontrol.s3-us-west-2.amazonaws.com/master/QGroundControl64.apk

PR #9238 (2020-12-06): Android Target SDK 28->29
Fix #9211
closed 2020-12-06 by DonLakeFlyer