mavlink/qgroundcontrol #9215: Cube Orange PX4 switch fails with "Unable to find specified firmware for board type"

dell640, 2020-11-26:
[As per discussion at px4.io](https://discuss.px4.io/t/cube-orange-px4-stack-issue/19754/7) with @DonLakeFlyer I'm rising this as issue:

I would like to upload PX4 stack to my Cube Orange, unfortunately without success.

When I upgrade my Cube to PX4 via MissionPlanner, upload is successful, however Cube hangs in bootloader and stack is never loaded and running.

When I try to upgrade to PX4 from QGC:
[image]
I’m getting this error:
[image]

I have two Cube Orange units here, both behaves the same.
Testing computer running W10.
Tried multiple QGC versions from 3.5.6 up to the latest and also daily and all have the same “error”.
Also I tried to uninstall USB drivers via Device Manager, that does not help. Then reinstalled UAV drivers, but problem persist.
Tried another computer (also W10) with same results.

DonLakeFlyer, 2020-11-27:
Turn on console logging for FirmwareUpgradeLog as described here and then attach the log so I can look at it.

DonLakeFlyer, 2020-11-27:
Also wait a minute these screenshots are not from a Daily build. They are from Stable. You need to use a daily build to get Cube Orange fixes.

DonLakeFlyer, 2020-11-27:
> Turn on console logging for FirmwareUpgradeLog as described here and then attach the log so I can look at it.

Never mind I can repro it myself.

DonLakeFlyer, 2020-11-28:
This is because of the ArduPilot bootloader on the board returning a different board id. Trying to figure out how to make it work...

dell640, 2020-11-28:
> 
> 
> This is because of the ArduPilot bootloader on the board returning a different board id. Trying to figure out how to make it work...

Thanks for checking that. As stated at first post, when proper PX4 firmware (v1.11.2 - cubepilot_cubeorange_default.px4) is uploaded via MissionPlanner (where is goes smoothly) Cube hangs in bootloader and PX4 stack is not running properly. So, maybe it is deeper incompatibility issue.
Looking forward to your discoveries.

DonLakeFlyer, 2020-11-28:
> As stated at first post, when proper PX4 firmware (v1.11.2 - cubepilot_cubeorange_default.px4) is uploaded via MissionPlanner (where is goes smoothly) Cube hangs in bootloader and PX4 stack is not running properly. So, maybe it is deeper incompatibility issue.

You should report that as a firmware issue. Not here.

Dule-sat, 2020-11-28:
[qgc issue cubeorange.txt](https://github.com/mavlink/qgroundcontrol/files/5611283/qgc.issue.cubeorange.txt)
Hi, I am uploading console log from QGC regarding the issue with Cube Orange. Like many, I have an issue when trying to flash Px4 via QGC.

Please help
Thanks

sibujacob, 2020-12-02:
As far as I know Pixhawke Orange CUBE has manufacturing defects especially the ones from PROFICNC.                                            They are not quality testing the product  for any defects .

FlyingPineapple, 2020-12-10:
> As far as I know Pixhawke Orange CUBE has manufacturing defects especially the ones from PROFICNC. They are not quality testing the product for any defects . There should be a program to test the product for all its port and POST functions before dispatch.

Do you have more information on this defects? Or where I can find more about it? 

I am also unable to get PX4 on a Cube Orange :( I tried both stable and daily builds and get the error message.

sibujacob, 2020-12-10:
> Do you have more information on this defects? Or where I can find more about it?
>
> I am also unable to get PX4 on a Cube Orange :( I tried both stable and daily builds and get the error message.

I am waiting for my Pixhawke Orange with ADS-B receiver to arrive in India.

DonLakeFlyer, 2020-12-10:
Fixed

closed 2020-12-10 by DonLakeFlyer

Dule-sat, 2020-12-14:
> Fixed

Can you please be more specific? Is it fixed with new daily build or master?

sibujacob, 2020-12-14:
> Fixed

what was the issue ...

Dule-sat, 2020-12-14:
It works with new daily release of QGC! All good now!

sibujacob, 2020-12-14:
Please share the link here

DonLakeFlyer, 2020-12-17:
A simple internet search on "qgroundcontrol daily build" will get you what you want

arunabha42, 2021-08-28:
> I am having the same issue with Cube orange for PX4 firmware load. Qgroundcontrol Daily Build does not solve the issue. I even tried QGC daily build in windows and Linux both system. uploaded the firmware using `make` command but no luck. Seems like cube orange is stuck in the bootloader loop. Only works with Arducopter firmware.

Can confirm, I'm facing the same issue. I'm on Windows 10.

Initially I was getting the "Unable to find specified firmware for board type", but after a few tries of jumping around installing ArduPilot and trying out QGC & QGC daily builds it reached the firmware installation stage.

In my case as well the orange light on the Cube Orange is always solid when the PX4 stack is installed. This is not the case when the ArduPilot stack is installed, the light goes off once it's booted up.

Looks like the firmware is loading onto the board but is not booting up. However, this is the case only with the PX4 stack, not when ArduPilot is being installed.

Any help appreciated I'm adding more details in #9843, thanks!

cph-devnet, 2021-12-07:
I was able to get my Cube Orange to flash with PX4, I just had to uninstall all windows drivers first.