mavlink/qgroundcontrol #9226: QGroundControl.AppImage not detecting joystick inside docker

anthonypan08, 2020-11-30:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
When I run `./QGroundControl.AppImage --logging:full 2>&1 | grep JoystickLog` it should detect my joystick

## Current Behavior
Please describe what happens instead of the expected behavior.
Right now it says 
```
JoystickLog: Available joysticks
JoystickLog: 	none found
```

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Started docker as described [here](https://dev.px4.io/master/en/test_and_ci/docker.html#calling-docker-manually)
2. Downloaded AppImage as described [here[(https://docs.qgroundcontrol.com/master/en/getting_started/download_and_install.html#ubuntu)
3. `./QGroundControl.AppImage --logging:full 2>&1 | grep JoystickLog`

I don't think this is a hardware issue because when I build from [source]( https://dev.qgroundcontrol.com/master/en/getting_started/index.html
) inside the docker (steps below) it is able to find the controller

1. git clone https://github.com/mavlink/qgroundcontrol.git --recursive
2. cd qgroundcontrol
3. git submodule update
4. sudo apt install -y speech-dispatcher libudev-dev libsdl2-dev
5. sudo apt install -y software-properties-common
6. sudo add-apt-repository ppa:beineri/opt-qt-5.12.6-bionic
7. sudo apt-get update
8. sudo apt install -y qt512-meta-full qt512charts-no-lgpl
9. cd build
10. source /opt/qt512/bin/qt512-env.sh
11. cmake ..
12. make
13. ./QGroundControl --logging:full 2>&1 | grep JoystickLog

It generates this 
```
JoystickLog: 	 "Logitech Gamepad F310" axes: 6 buttons: 15 hats: 0 isGC: true
JoystickLog: _loadSettings  "Logitech Gamepad F310"
JoystickLog: _loadSettings calibrated:txmode:throttlemode:exponential:deadband:badsettings true 2 1 0 false false
JoystickLog: _loadSettings axis:min:max:trim:reversed:deadband:badsettings 0 -32767 32767 0 false 0 false
JoystickLog: _loadSettings axis:min:max:trim:reversed:deadband:badsettings 1 -32767 32767 0 true 0 false
JoystickLog: _loadSettings axis:min:max:trim:reversed:deadband:badsettings 2 -32767 32767 0 false 0 false
JoystickLog: _loadSettings axis:min:max:trim:reversed:deadband:badsettings 3 -32767 32767 0 true 0 false
JoystickLog: _loadSettings axis:min:max:trim:reversed:deadband:badsettings 4 -32767 32767 0 false 0 false
JoystickLog: _loadSettings axis:min:max:trim:reversed:deadband:badsettings 5 -32767 32767 0 false 0 false
JoystickLog: _loadSettings function:axis:badsettings 0 2 false
JoystickLog: _loadSettings function:axis:badsettings 1 3 false
JoystickLog: _loadSettings function:axis:badsettings 2 0 false
JoystickLog: _loadSettings function:axis:badsettings 3 1 false
JoystickLog: _loadSettings function:axis:badsettings 4 4 false
JoystickLog: _loadSettings function:axis:badsettings 5 5 false
JoystickLog: Available joysticks
JoystickLog: 	Skipping duplicate "Logitech Gamepad F310"
```

It also shouldn't be the AppImage because I ran it outside the docker and it was able to detect the hardware. My best guess right now is that somehow some docker permission is preventing the AppImage from picking up the hardware (but weirdly building from source bypassed that permission)

patrickelectric, 2020-12-02:
You prob need to bind the /dev with the docker container.

DonLakeFlyer, 2020-12-02:
If you can reproduce outside of the docker image then I'll buy it's a QGC problem. But without that it seems like docker problem. I'm not hearing any reports from other folks about Joystick problems on Linux.

closed 2020-12-02 by DonLakeFlyer

anthonypan08, 2020-12-03:
> You prob need to bind the /dev with the docker container.

Just for the record /dev didn't seem to work for me.
I did `docker-compose run -v /dev:/dev my-docker bash`

anthonypan08, 2020-12-03:
Also @DonLakeFlyer my issue seem to be the same as [9041](https://github.com/mavlink/qgroundcontrol/issues/9041). Do you know if/how that was solved

DonLakeFlyer, 2020-12-03:
> Do you know if/how that was solved

Nope.