mavlink/qgroundcontrol #9236: Current Daily: terrain altitude bar seems wrong

Antiheavy, 2020-12-05:
Current daily build seems to have something going on with the terrain display.  I'm testing indoors with a vehicle connected and no GPS fix (or home location yet).  QGC seems to understand the local position ground altitude (as shown in the Mission Start), however, it is showing the mission plan as being completed below the terrain.

Screenshot:
[image]

Also: does the RED flight path lines indicate a problem, in this case below terrain?

DonLakeFlyer, 2020-12-05:
Yeah, something screwed up there. I thought I already had a 4.1 issue on it somewhere but can't find now. Can you attach the plan you are testing with.

The red line indicates below terrain.

Antiheavy, 2020-12-05:
>  Can you attach the plan you are testing with.

couple examples attached in this zip folder.
[below-ground-missions.zip](https://github.com/mavlink/qgroundcontrol/files/5647686/below-ground-missions.zip)

moreba1, 2020-12-07:
1- when we click on "SET TO MAP CENTER" button home altitude set to 0.
2- when we change location of Launch with mouse (home) , altitude of other relative waypoint does not changed .

DonLakeFlyer, 2020-12-07:
> 1- when i click on "SET TO MAP CENTER" button home altitude set to 0. also in end of plan file altitude of "plannedHomePosition": is 0.
> 2- when i change location of Launch with mouse (home) , altitude of other relative waypoint does not changed .

Can you put these in a separate issue?

moreba1, 2020-12-07:
> 
> 
> > 1- when i click on "SET TO MAP CENTER" button home altitude set to 0. also in end of plan file altitude of "plannedHomePosition": is 0.
> > 2- when i change location of Launch with mouse (home) , altitude of other relative waypoint does not changed .
> 
> Can you put these in a separate issue?

these are reason of terrain altitude bar issue.

DonLakeFlyer, 2020-12-07:
> these are reason of terrain altitude bar issue.

So far, not according to the plan files attached to this issue. They have good launch altitudes in them. I think this a separate issue, possibly related somehow but maybe not.

DonLakeFlyer, 2020-12-07:
Wait I looked inside the plan file and the `"AMSLAltAboveTerrain": null` field is a NaN which is incorrect. So possible caused by 0 height launch position but not sure how a NaN would get in there. More likely caused by a save being allowed prior to terrain heights being downloaded. Still, there are multiple bugs here. Keeping them in separate issue is easier to track individual problems. Dragging a launch position and having it screw up the alt is a separate bug.

DonLakeFlyer, 2020-12-07:
@antiheavy When you created these plans did you move the Launch Position?

DonLakeFlyer, 2020-12-07:
> More likely caused by a save being allowed prior to terrain heights being downloaded.

Nope. This is a relative alt mission. Terrain isn't needed to save.

DonLakeFlyer, 2020-12-07:
@moreba1 Ok, digging further the plan file is incorrect in that it has a persisted alt of 0 for planned home. So it is likely only the dragging problem. Sorry going to fast and missed this part of your comment "also in end of plan file altitude of "plannedHomePosition": is 0."

DonLakeFlyer, 2020-12-07:
> I'm testing indoors with a vehicle connected and no GPS fix (or home location yet).

@Antiheavy I think there still might be a bug related to this specific test case. Can you retest after this current fix goes in?

moreba1, 2020-12-07:
maybe related to this change:
[image]

DonLakeFlyer, 2020-12-08:
I'll reopen if the vehicle connected with no GPS causes problems. Looking at the code again I don't think so since the vehicle will not broadcast HOME_POSITION in that case.

Antiheavy, 2020-12-09:
Sorry for the delayed reply, yes I probably dragged the launch position. I often do that out of habit.

Antiheavy, 2020-12-09:
@DonLakeFlyer I;ve been testing today's daily and I cannot seem to repeat the same error.  Is this expected based on improvements you made?   However, I'm planning a mission over an area with rolling hills but I don't see any rolling hill terrain info display along the bottom bar:
[image]

DonLakeFlyer, 2020-12-10:
> I;ve been testing today's daily and I cannot seem to repeat the same error. Is this expected based on improvements you made?

Yes.

> However, I'm planning a mission over an area with rolling hills but I don't see any rolling hill terrain info display along the bottom bar:

Terrain data resolution is 1-arc second which is about 100 feet. So you might now see any deviation over small distances or small elevation change.

Antiheavy, 2020-12-11:
> Terrain data resolution is 1-arc second which is about 100 feet. So you might now see any deviation over small distances or small elevation change.

the golf course shown above has rolling hills all across it, maybe 150 feet vertical variation from lowest to highest spots across multiple hills.  

When I enable the follow terrain checkbox you can see that the hills are there, I would expect to see the "ground" line on the lower part of the altitude profile to show this same variation.  plus all the waypoints look crammed down at "0" altitude which also looks wrong/confusing.

[image]

DonLakeFlyer, 2020-12-11:
Yup, screwed up. Can you attach that plan. I can use it for testing.

> plus all the waypoints look crammed down at "0" altitude which also looks wrong/confusing.

That's on purposed. The circular indicators don't track the height they stay on the bottom. This way you can see the flight path and terrain lines unobscured.

Antiheavy, 2020-12-11:
> Yup, screwed up. Can you attach that plan. I can use it for testing.

the mystery deepens:  i fired up QGC again to try to recreate and save a mission showing the issue, however, since my autopilot was still plugged into my PC it auto-connected and downloaded the onboard mission plan, which happens to be the same plan as shown above, but now it's all red lines and no terrain info.  screenshot and saved mission plan below:
[image]
[mission plan downloaded off autopilot and now terrain and altitude looks weird.zip](https://github.com/mavlink/qgroundcontrol/files/5681416/mission.plan.downloaded.off.autopilot.and.now.terrain.and.altitude.looks.weird.zip)

DonLakeFlyer, 2020-12-19:
The plan files from the above comment will now load "better" with this last set of changes. FW Landing and Takeoff flight profiles are still not quite right.

Antiheavy, 2020-12-19:
here is a newly planned mission using today's daily.   Terrain follow enabled.  the ground elevation display seems scaled incorrectly or some other problem, it doesn't follow the mission undulations like you'd expect.  screenshot here and saved 
[mission and terrain undulations do not match.zip](https://github.com/mavlink/qgroundcontrol/files/5720226/mission.and.terrain.undulations.do.not.match.zip)
mission plan attached.
[image]

DonLakeFlyer, 2020-12-19:
That is because prior to #9265 the terrain profile min/max altitude calc was incorrect. Also the terrain profile was clipped to the min height of the chart display. So that flat lines of terrain are actually incorrect and are just being clipped at the bottom. The pull I just merged a few minutes ago should fix all this. Will take an hour or so for a new daily to pop out.

Antiheavy, 2020-12-20:
> merged a few minutes ago

oops!  I thought the version I was testing already had the fix.  I re tested with the newer daily and it's looking better (however, as you say, still something goofed with takeoff, it's showing red here):
[image]

moreba1, 2020-12-20:
> Terrain data resolution is 1-arc second which is about 100 feet.

Can be displayed more resolution of terrain in takeoff and landing area?

Antiheavy, 2020-12-20:
> > Terrain data resolution is 1-arc second which is about 100 feet.
> 
> Can be displayed more resolution of terrain in takeoff and landing area?

The resolution of the source data is the limiting factor.  I suspect SRTM (Shuttle Radar Topography Mission) is the original source data which is about 1 arc-second (~30 meters), most elevation map providers use this is as their main layer since it was given to the world for free and has nearly full globe coverage.  https://en.wikipedia.org/wiki/Shuttle_Radar_Topography_Mission

DonLakeFlyer, 2020-12-20:
> Can be displayed more resolution of terrain in takeoff and landing area?

Don't quite understand what exactly you are asking for?

DonLakeFlyer, 2020-12-30:
This should be all fixed up now

closed 2020-12-30 by DonLakeFlyer