mavlink/qgroundcontrol #9244: TransectStyleComplexItem can report ready while not ready

remster, 2020-12-08:
## Expected Behavior
when `TransectStyleComplexItem::readyForSaveState() == VisualMissionItem::ReadyForSave` the `_transects` should have terrain values set.

## Current Behavior
It may happen that it is not true because in [this](https://github.com/mavlink/qgroundcontrol/blob/master/src/MissionManager/TransectStyleComplexItem.cc#L601) bit of code, that today reads:
```
        emit readyForSaveStateChanged();

        // Now that we have terrain data we can adjust
        _adjustTransectsForTerrain();
```
and so the signal that informs that `readyForSaveState` may have just become `VisualMissionItem::ReadyForSave`, somebody may inquire that state and discover that it is, because `terrainReady = _transectsPathHeightInfo.count();` is non-zero [here](https://github.com/mavlink/qgroundcontrol/blob/master/src/MissionManager/TransectStyleComplexItem.cc#L624)., while the transects have not yet been adjusted for terrain. 

## Steps to Reproduce:
Unsure - reproduces in a test environment in a custom build that is hard to supply.

## System Information
When posting bug reports, include the following information
- QGC Version: 4f32b8d46a1c07b4a82e166463d8a7ffc7a1040b

DonLakeFlyer, 2020-12-08:
Thanks for the report. I"m re-writing that code right now to support terrain adjustment in the turn segments. So I'll fix it.

DonLakeFlyer, 2020-12-08:
Related to #8220

DonLakeFlyer, 2020-12-29:
Yeah, the signal going out is a bit too early. Thanks for the report.

PR #9286 (2020-12-30): TransectStyleComplexItem - Fix terrain ready signaling
* Terrain ready was being signaled too early
* Fix #9244
closed 2020-12-30 by DonLakeFlyer