mavlink/qgroundcontrol #9247: show vehicle icon above waypoints, lines and trajectory

moreba1, 2020-12-11:
When vehicle fly in a busy trajectory and close waypointS (in survey) it is hard to detect vehivle location .

Antiheavy, 2020-12-12:
oh yeah... that is quite annoying.  I could see why you might want the waypoints on top in Plan view, but in Fly view you probably want the vehicle on top?  Unless that causes problems selecting waypoints or performing other actions?
[image]

moreba1, 2020-12-12:
yes , there is need to see vehicle on of all items.

Antiheavy, 2020-12-12:
Looking at older versions of QGC it seems the vehicle icon has always been behind the mission items.  I wonder if this topic has come up before and there might be reasons it's shown below.  @DonLakeFlyer do you know?

DonLakeFlyer, 2020-12-12:
It's so you can click on waypoints to change current item.

moreba1, 2020-12-12:
> 
> 
> It's so you can click on waypoints to change current item.

in Fly view, mostly waypoint that we will select (for go to waypoint) have distance from vehicle , what about for waypoint lines and trajectory ?

DonLakeFlyer, 2020-12-12:
"mostly" works for you. But "mostly" never works for QGC as a whole.

Creating click targets for waypoint line would be a nightmare. Especially on mobile where the click targets need to be expanded to make them work.

This problem has been exacerbated by somewhere along the way the vehicle icon being made semi-opaque. Not sure why that was done. I'm unconvinced that was a good change.

moreba1, 2021-09-20:
> "mostly" works for you. But "mostly" never works for QGC as a whole.
>
> Creating click targets for waypoint line would be a nightmare. Especially on mobile where the click targets need to be expanded [...]
can we show vehicle icon above all items except waypoints ? 
How can i do it for myself ? i edited and added zorder of vehicle icon (in code) but it did not worked.

closed 2026-10-04 by github-actions[bot]