mavlink/qgroundcontrol #9250: When "High Latency" is checked on a connection, QGC still loads parameters

vorobotics, 2020-12-13:
When "High Latency" is checked, QGC still loads parameters upon connection. 
I am using QGC 4.0.11 from GitHub releases page on Ubuntu Linux 18.04.3. 

It looks like this could maybe be remedied by replacing this line:
https://github.com/mavlink/qgroundcontrol/blob/ca03468333203e27125759bb59aaca4b299019da/src/Vehicle/InitialConnectStateMachine.cc#L267

With the following code:
```(cpp)
SharedLinkInterfacePtr sharedLink = weakLink.lock();
if (sharedLink->linkConfiguration()->isHighLatency()) {
  qCDebug(InitialConnectStateMachineLog) << "_stateRequestParameters: Skipping first parameter load request due to link type";
  // do nothing else
} else {
  qCDebug(InitialConnectStateMachineLog) << "_stateRequestParameters";
  vehicle->_parameterManager->refreshAllParameters();
}
```

DonLakeFlyer, 2020-12-13:
Current stable is longer being changed. New stable is being worked on. Can you retry with daily build?

vorobotics, 2020-12-13:
Yes, this issue still happens with daily build. When "high latency" is checked, it still loads the parameter list. 

The version tested:
`HEAD:ca0346833 2020-12-11 16:17:08 -0800`

DonLakeFlyer, 2020-12-18:
Not downloading parameters is working fine for me. ParameterManager handles it: https://github.com/mavlink/qgroundcontrol/blob/master/src/FactSystem/ParameterManager.cc#L454.

What is not working for me is that GeoFence/Rally point download is attempted.

DonLakeFlyer, 2020-12-18:
I'll re-open if you can give me a repro which fails.

closed 2020-12-18 by DonLakeFlyer