mavlink/qgroundcontrol #9255: Joystick channels 5/6 not working

npyoung, 2020-12-16:
I am using an Xbox One controller with QGC to control a quad running Arducopter. I have gimbal control enabled. I can calibrate all 6 axes on the control (two joysticks and two triggers). All is well with the first four axes: throttle/pitch/roll/yaw (the sticks).

I set up channel 5 to control a pitch-only servo gimbal, which should be mapped to one of my triggers. When I look at the calibration screen, I can see the slider moving with my triggers. However on the main Joystick screen, touching the trigger causes the slider circle to just... disappear? Until I release it and it reappears. The triggers have no effect on the gimbal, but I know the servo is set up properly because I can put the gimbal in stabilize mode and it will compensate for quad attitude using the gyros.

[image]

## System Information
- Operating System: MacOS High Sierra
- QGC Version: 4.0.11
- QGC build: stable
- Flight Controller: Matek F405-CTR
- Autopilot (with version): Arducopter 4.0.5

DonLakeFlyer, 2020-12-16:
As far as I know ArduPilot doesn't yet support the new Mavlink gimbal protocol which is what QGC supports. So this isn't going to work on ArduPilot. If you want that functionality you should request it from ArduPilot folks. If you find that ArduPilot says they've added support and it still doesn't work let me know and I'll re-open.

npyoung, 2020-12-16:
I don't think I'm talking about the mavlink gimbal protocol. The typical setup for that is a separate gimbal controller that gets some kind of serial input from the FC. What I'm trying to do is control a simple PWM servo with a 5th axis on a joystick.

DonLakeFlyer, 2020-12-16:
You might not be talking about that, but that is how QGC gimbal support is meant to work. You are using an ArduPilot setup which is meant to be controlled through an RC transmitter channel. QGC joystick does not simulate an RC such that you could use it to drive an RC PWM driven servo control.

> Also QGC's GUI is misbehaving here with the disappearing slider. At the very least that seems like it can't be an Ardupilot issue.

Yup, some bug there. That said, now that I remember, I don't think any of the Gimbal control ui stuff is supposed to be showing up in the build. I don't think it was ever finished. So the likely fix is to just hide it.

DonLakeFlyer, 2020-12-16:
I'm leaving this open for now to look at what the state of gimbal is. Might need to remove it for Stable 4.1

DonLakeFlyer, 2020-12-29:
None of this gimbal stuff works at all. There is just some ui fluff implemented which goes nowhere.

DonLakeFlyer, 2020-12-30:
Removed non-working code/ui

PR #9285 (2020-12-30): Joystick: Remove incomplete gimbal code
* There was a bunch of UI code related to gimbal joystick setup which did nothing other than confuse users.
* This pull can be reverted if anyone wants to resurrect.
* Not sure how something which completely didn't work go merged into the repo in the first place
* Fix #9255
closed 2020-12-30 by DonLakeFlyer