mavlink/qgroundcontrol #9260: Segfault on vehicle reboot on master

bkueng, 2020-12-17:
Happened to me once during a vehicle reboot connected via USB. I didn't try much to reproduce.
```
Temp log "/tmp/FlightData893795.mavlink"
LinkManager::sharedLinkInterfaceForLink returning nullptr
./staging/qgroundcontrol-start.sh: line 10: 50355 Segmentation fault      (core dumped) "${HERE}/QGroundControl" "$@"
```

DonLakeFlyer, 2020-12-17:
Any more details? You rebooted how? From Parameter Editor? And then did nothing at all after and then it crashed?

bkueng, 2020-12-18:
> From Parameter Editor?

Yes, and it crashed directly after, no other interaction.

DonLakeFlyer, 2020-12-18:
What kind of board? Doubt it matters but I tried to repro and it worked fine of course.

bkueng, 2020-12-21:
A Pixhawk 4

PR #9267 (2020-12-27): Fix Segfault on vehicle reboot
When rebooting the vehicle from param editor, the vehicle would call `VehicleLinkManager::closeVehicle()` from `Vehicle::_mavlinkMessageReceived`. The link was then removed, and deleted as the ref count went to 0. If `MAVLinkProtocol::receiveBytes` was then parsing further messages, the same link was accessed again, leading to invalid memory access.
This PR fixes it by checking the validity of the link object after message handling. It could be solved differently, e.g. by closing the connection outside of the `MAVLinkProtocol::receiveBytes` call.

See individual  [... cut]
closed 2020-12-27 by DonLakeFlyer