mavlink/qgroundcontrol #9263: Crash referencing stale _activeVehicle in Joystick code

DonLakeFlyer, 2020-12-18:
Crash is here: https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/Vehicle.cc#L3804 caused by referencing a stale _activeVehicle in the Joystick code.

It hit this using PX4 SITL:
* Start sitl with an iris
* Start QGC with joystick enabled
* Stop SITL
* QGC shows comm lost
* Click Disconnect
* Boom

It's some sort of race condition. I can't get it to reproduce now. But from looking at the joystick code _activeVehicle is set in Joystick::startPolling. But when a vehicle is being destructed I don't see how that _activeVehicle gets cleared. I'm not familiar with this code at all but something seems wrong here with respect to there not being a stopPolling in the Vehicle code.

DonLakeFlyer, 2020-12-18:
@jaxxzer Can someone from your team look at this. Likely faster since I'm not familiar with the joystick code any more.

jaxxzer, 2021-01-14:
@DonLakeFlyer  we are trying to replicate it here.

DonLakeFlyer, 2021-01-14:
I can't make it happen again myself. Will likely need to be tracked down through code inspection. When looking through code myself something seemed fishy as I described above. But I don't know that code enough to say exactly what. Smells wrong through.

closed 2021-02-17 by DonLakeFlyer