mavlink/qgroundcontrol #9264: VTOL can not do vtol_land in landing pattern

moreba1, 2020-12-19:
In my SITL test with v1.11.1 px4 and standard vtol airframe (weather wane disabled) , VTOL can not do VTOL land in landing pattern.

reason of this issue is that param_4 of MAV_CMD_NAV_VTOL_LAND is 0 and should be Nan.

DonLakeFlyer, 2020-12-19:
Did something change here on the firmware side I'm pretty sure that used to work? Without and altitude specified how do you land somewhere which is at a different alt than the home position?

moreba1, 2020-12-20:
Param4 is for set Yaw not Alt.

DonLakeFlyer, 2020-12-22:
Oops, yeah my bad.

DonLakeFlyer, 2020-12-28:
@moreba1 A new daily should be out in an hour or so. Could you verify? I'm having problems with VTOL SITL for some reason I don't understand yet.

closed 2020-12-28 by DonLakeFlyer

moreba1, 2020-12-28:
yes, it is correct.

PR #9279 (2020-12-28): Fix VTOL Landing yaw param
Fix #9264