mavlink/qgroundcontrol #9273: daily build: terrain intersect warning too sensitive for fixed wing takeoff

Antiheavy, 2020-12-24:
I suspect that the terrain intersect logic for the transect from Launch to Takeoff points needs some special handling to avoid terrain intersections being calculated very close to the Launch location.

Example:
this shows no terrain intersections:
[image]

but moving takeoff slightly shows a terrain intersection (red line):
[image]

The actual terrain near the Launch location is relatively flat, but there is likely a tiny rise or step (less than 1 meter) in terrain elevation just to the north of the Launch.  If Launch is calculated to be exactly at whatever the terrain database says then there may easily be "false positive" terrain intersections very near to the Launch location.  I've run into this in the past when developing/testing other ground control station applications which is why I suspect something similar here.

Suggestions:   
- Ignore terrain intersections within a 10 meter radius "buffer" around the Launch location.  Document this somewhere.
- Or, assume "Launch" is 1 meter above terrain (seems more hacky)
- Or, some combinations of approaches

DonLakeFlyer, 2020-12-27:
Hmm, yeah...

DonLakeFlyer, 2020-12-29:
This was a bug in the collision detection code. You can still theoretically get a collision indication a the beginning on the takeoff segment on hilly terrain that may or may not be false. But with the current code I believe it should indicate an "iffy" situation to fly in as opposed to just being wrong. 

Will need some bake time to determine whether doing something special is still required as suggested above. If folks are seeing it giving misleading results we can update it then. For now I'd rather err to the side of caution in warning possibly incorrectly in extreme cases.

PR #9284 (2020-12-30): Terrain Profile: Fix off by 1 on terrain collision detection
* This could cause FW takeoff to show incorrect collision right at takeoff point
* Fix #9273
closed 2020-12-30 by DonLakeFlyer