mavlink/qgroundcontrol #9275: Issue with download function of FTPManager class

1AwesomeDeveloper, 2020-12-26:
I have installed the latest version of QGC from this page, And I was trying to use the FTPManager class in QGC for downloading a specific file from drone SD card to my system via Mavlink. As I see in the FTPManager class we have a download function that can be used to download a file via mavlink. But when I try to download a file, it doesn't get downloaded. Sometimes it shows me the file in my pc that I intend to download, but when I open that files, it says unable to open. 
Can you please verify that the download function in FTPManager class, is performing all the steps in correct manner ?

DonLakeFlyer, 2020-12-27:
As far as I know FTP download works fine since it is used by the new `COMPONENT_INFORMATION` code which is tested with unit tests. When you say "latest version" I don't know what that means. If you are using master it should work. If you are having problems first step is to turn on logging for FTPManager to see what is happening. There is possibly bugs but without detailed information on logging to show the error there is no way for me to prove that. I will re-open if you can provide detailed repro steps with logs showing the error.

closed 2020-12-27 by DonLakeFlyer

1AwesomeDeveloper, 2020-12-28:
I have a key.txt in my Pixhawk SD card and I want to download it, using the download function in FTP Manager class, 
here is how I have used it 
[image]

I used in MultiVehicleManager.cc because the constructor of FTPManager requires an object of Vehicle class..

Here is the Logged output,
[image]

yes, I am using the master branch.

DonLakeFlyer, 2020-12-28:
The log output tells you exactly what is happening. The vehicle side did not respond to the Open file request and QGC times out waiting for the response. Nothing QGC can do about that. You'll have to figure out why the Vehicle didn't respond.

LorenzMeier, 2020-12-29:
The URL of the file is incorrect - any file on the SD card is on /fs/microsd/key.txt, NOT /key.txt

1AwesomeDeveloper, 2020-12-29:
[image]
QGC has a FTP Manager test class that I was following here the file paths are given as /version.json.

1AwesomeDeveloper, 2020-12-29:
Sir, I really think there is an issue with naming of directories, (How QGC uses for FTP and how PX4 uses in Mavlink console). Kindly clarify.

LorenzMeier, 2020-12-29:
No there is not, because your assumption that the file `version.json` is located on the SD card is incorrect. I can see from your screenshot that you still have the wrong path. You are using `//fs/microsd`, you should be using `/fs/microsd`. It is clearly visible in the log.

I would appreciate if you could stop implying that the system is broken.

1AwesomeDeveloper, 2020-12-29:
Sorry sir, for my claim that the system is broken, you can check now the path is correct now (line 132), earlier I was using mftp:// before the actual path 
[image].
Could you please look into it

1AwesomeDeveloper, 2020-12-29:
[image]

To the best I can understand, the QGC strips off everything else except filename (line 65). And even I pass my fully Qualified path(/fs/microsd/key.txt) for SD card, it doesnot get to know from where to download it. If wrong then please correct.
 Thank you.

DonLakeFlyer, 2020-12-29:
> To the best I can understand, the QGC strips off everything else except filename (line 65).

Yes, this code strips off everything but the filename. And no, this specific variable is not what is sent to the vehicle. `_downloadState.fullPathOnVehicle` is what is sent to the vehicle which is the fully qualified path as can be seen in the log output.

1AwesomeDeveloper, 2020-12-31:
Sir, I cloned current master and build it without changing any code, when I try to access Mavlink console from QGC it says, that Mavlink console provides connection to system, and It doesn't reponds to any of my commands, can you please check this. I am writing this problem here because I think its somewhat related to my FTP problem because I am unable to access the filesystem using console also,

Pardon if I am wrong