mavlink/qgroundcontrol #9277: RNG1FND_TYPE list do not contain HR04 (30) list item

mrRobot62, 2020-12-27:
## Expected Behavior
I try to install a HC04 sensor on my Pixhawk 2.4.8 (Rover 4.0 FW). Inside listbox of RNGFND1_TYPE I do not see an appropriate value. If I try to insert manually 30 (for HC04) I got this:

## Current Behavior
[image]

Following types missing:
[image]

## Steps to Reproduce:
GC-Control 4.0.11 (on Mac), go to all parakeets. Fetch RNGFND1_TYPE - no entry for HC04 available

## System Information
MAC OS 10.15.7 (Catalina)
QGC 4.0.11
Pixhawk 2.4.8
Ardupilot-Rover 4.0

DonLakeFlyer, 2021-01-02:
Parameter metadata is out of date. Also FYI: It is fine to set this manually like you do. That will work.

DonLakeFlyer, 2021-01-02:
Not going to hold stable for this

DonLakeFlyer, 2021-07-20:
I'm updating the parameter metadata but Rover 4.0 fw sources do not have an updated RNGFND1_TYPE. Only 4.1 has it updated.