mavlink/qgroundcontrol #9290: Code not compiled on latest Android SDK and NDK.

Dhananjay-Khairnar, 2020-12-31:
I have following setup 
Host Machine : Windows 10
Target System : Android 7
NDK version : 21
API level : 25

I am trying to compile code on Qt creator for Android device got following error.
clang++: error: no such file or directory: '@armeabi-v7a\object_script.staging\libQGroundControl_armeabi-v7a.so.Armeabi-v7a'

### Log
16:35:17: Running steps for project qgroundcontrol...
16:35:17: Starting: "C:\Qt\Qt5.14.1\5.14.1\android\bin\qmake.exe" C:\Users\Owner\Downloads\qgroundcontrol\qgroundcontrol.pro -spec android-clang "CONFIG+=debug" "CONFIG+=qml_debug" CONFIG+=installer ANDROID_ABIS="armeabi-v7a"
'which' is not recognized as an internal or external command,
operable program or batch file.
C:\Qt\Qt5.14.1\5.14.1\android\bin\rcc.exe: File does not exist 'C:/qmake_qmake_qm_files.qrc'
'which' is not recognized as an internal or external command,
operable program or batch file.
Project MESSAGE: Qt version 5.14.1
Project MESSAGE: Android Arm 64 bit build
Project MESSAGE: Daily Build
Project MESSAGE: QGroundControl Development master:dacd28b66 2020-12-30 10:22:34 -0800
Project MESSAGE: Build Installer
Project MESSAGE: Release flavor
Project MESSAGE: Using Default QtLocation headers
Project MESSAGE: Skipping support for Pairing
Project MESSAGE: Using MAVLink dialect 'ardupilotmega'.
Project MESSAGE: ANDROID_EXTRA_LIBS
Project MESSAGE: arm64-v8a
Project MESSAGE: C:/Users/Owner/Downloads/qgroundcontrol/libs/OpenSSL/android_openssl/arm64/libcrypto.so C:/Users/Owner/Downloads/qgroundcontrol/libs/OpenSSL/android_openssl/arm64/libssl.so
Project MESSAGE: Skipping support for Zeroconf (unsupported platform)
Project MESSAGE: Looking for Airmap in folder "C:/Users/Owner/Downloads/qgroundcontrol/libs/airmapd", variant: "Qt.5.14"
Project MESSAGE: Skipping support for Airmap (unsupported platform)
Project MESSAGE: Taisync disabled
Project MESSAGE: Microhard disabled
Project MESSAGE: Including support for video streaming
Project MESSAGE: Adding Serial Java Classes
Project MESSAGE: This project is using private headers and will therefore be tied to this specific Qt module build version.
Project MESSAGE: Running this project against other versions of the Qt modules may crash at any arbitrary point.
Project MESSAGE: This is not a bug, but a result of using Qt internals. You have been warned!
Project MESSAGE: Qt version 5.14.1
Project MESSAGE: Android Arm 32 bit build
Project MESSAGE: Daily Build
Project MESSAGE: QGroundControl Development master:dacd28b66 2020-12-30 10:22:34 -0800
Project MESSAGE: Build Installer
Project MESSAGE: Release flavor
Project MESSAGE: Using Default QtLocation headers
Project MESSAGE: Skipping support for Pairing
Project MESSAGE: Using MAVLink dialect 'ardupilotmega'.
Project MESSAGE: ANDROID_EXTRA_LIBS
Project MESSAGE: armeabi-v7a
Project MESSAGE: C:/Users/Owner/Downloads/qgroundcontrol/libs/OpenSSL/android_openssl/arm/libcrypto.so C:/Users/Owner/Downloads/qgroundcontrol/libs/OpenSSL/android_openssl/arm/libssl.so
Project MESSAGE: Skipping support for Zeroconf (unsupported platform)
Project MESSAGE: Looking for Airmap in folder "C:/Users/Owner/Downloads/qgroundcontrol/libs/airmapd", variant: "Qt.5.14"
Project MESSAGE: Skipping support for Airmap (unsupported platform)
Project MESSAGE: Taisync disabled
Project MESSAGE: Microhard disabled
Project MESSAGE: Including support for video streaming
Project MESSAGE: Adding Serial Java Classes
16:35:31: The process "C:\Qt\Qt5.14.1\5.14.1\android\bin\qmake.exe" exited normally.
16:35:31: Starting: "C:\Users\Owner\AppData\Local\Android\Sdk\ndk\21.3.6528147\prebuilt\windows-x86_64\bin\make.exe" -f C:/Users/Owner/Downloads/qgroundcontrol/build/Makefile qmake_all
make: Nothing to be done for 'qmake_all'.
16:35:32: The process "C:\Users\Owner\AppData\Local\Android\Sdk\ndk\21.3.6528147\prebuilt\windows-x86_64\bin\make.exe" exited normally.
16:35:32: Starting: "C:\Users\Owner\AppData\Local\Android\Sdk\ndk\21.3.6528147\prebuilt\windows-x86_64\bin\make.exe" -j4
rcc qgroundcontrol_qmlcache.qrc 
rcc src_FirmwarePlugin_APM_APMResources_qmlcache.qrc 
rcc src_FirmwarePlugin_PX4_PX4Resources_qmlcache.qrc 
rcc src_Airmap_dummy_airmap_dummy_qmlcache.qrc 
compiling armeabi-v7a\qrc_qgroundcontrol_qmlcache.cpp 
compiling armeabi-v7a\qrc_src_FirmwarePlugin_PX4_PX4Resources_qmlcache.cpp 
compiling armeabi-v7a\qrc_src_Airmap_dummy_airmap_dummy_qmlcache.cpp 
compiling armeabi-v7a\qrc_src_FirmwarePlugin_APM_APMResources_qmlcache.cpp 

linking staging\libQGroundControl_armeabi-v7a.so 
clang++: error: no such file or directory: '@armeabi-v7a\object_script.staging\libQGroundControl_armeabi-v7a.so.Armeabi-v7a'
make: *** [Makefile:2309: staging\libQGroundControl_armeabi-v7a.so] Error 1
16:35:41: The process "C:\Users\Owner\AppData\Local\Android\Sdk\ndk\21.3.6528147\prebuilt\windows-x86_64\bin\make.exe" exited with code 2.
Error while building/deploying project qgroundcontrol (kit: Android for armeabi-v7a,arm64-v8a,x86,x86_64 (Clang Qt 5.14.1 for Android))
When executing step "Make"
16:35:41: Elapsed time: 00:24.

kressinluiz, 2021-01-01:
Do you got it working with Qt 5.12.6 and ndk 19 as explained [here](https://dev.qgroundcontrol.com/master/en/getting_started/) ? Hope it helps

DonLakeFlyer, 2021-01-01:
If you want to see the details of how QGC builds Android you can look at `.travis.yml`. This way you can duplicate the NDK/SDK settings and so forth. Other than that getting Android + Qt setup correctly to build is always a royal PITA. Doubly so on Windows. I'm not sure I've heard anyone figuring that out yet. All of that said, this is 99% likely not a bug in QGC and a setup problem. Issue are for bugs.

closed 2021-01-01 by DonLakeFlyer