mavlink/qgroundcontrol #9292: Camera trigger not working in QGC missions.

michael-fxj, 2021-01-01:
## Expected Behavior
When adding a MAV_CMD_DO_DIGICAM_CONTROL ("Camera Control" in QGC) command with parameter five set to 1 in a mission the camera should be triggered when the command is executed in the air.

## Current Behavior
The camera is not triggered in a mission using MAV_CMD_DO_DIGICAM_CONTROL command, but it works using the "Trigger Camera" button in QGroundControl.

## Steps to Reproduce:
To test the camera trigger I used airsim hitl simulation and a raspberry pi to check for a trigger signal. Alternatively a multimeter could also be used and the trigger period set to 1000 ms.
1. Set the camera settings to:
Trigger mode: Time based on, on command
Trigger interface: GPIO
Trigger Pin Polarity: Low (0V)
Trigger Period: 40 ms
2. Create a simple mission in QGroundControl. Add one mission command "Camera control" with "Command" (Parameter 5) set to 1. For a working flight plan set "Command" to 106.5353216 (you cannot do this in QGC, you have to save the mission and edit the .plan file in order to do this).
Alternatively just open one of the following plan files (rename to .plan before using, git doesnt allow .plan files):
Not working plan:
[trigger_param1.txt](https://github.com/mavlink/qgroundcontrol/files/5759206/trigger_param1.txt)
Working plan with 106.5353216:
[trigger_param106.txt](https://github.com/mavlink/qgroundcontrol/files/5759207/trigger_param106.txt)
3. Use a raspberry pi, multimeter, oscilloscope or something else to catch the trigger signal. To verify it is working use the "Camera Trigger" button from QGC.
4. Upload one of the plans to the flight controller. It will only triggeer the camera with the described parameter set to 106.5353216.

## System Information
When posting bug reports, include the following information
- Operating System: Windows 10
- QGC Version: v4.0.11
- QGC build: stable
- Flight Controller: Pixhawk 4
- Autopilot (with version): PX4 1.11.2

## Detailed Description
First workaround I found was using the "Trigger Camera" button in QGC and it works fine.
The generated .plan has the following command included:

> {
>                 "autoContinue": true,
>                 "command": 203,
>                 "doJumpId": 4,
>                 "frame": 2,
>                 "params": [
>                     0,
>                     0,
>                     0,
>                     0,
>                     1,
>                     0,
>                     0
>                 ],
>                 "type": "SimpleItem"
> }

According to the [Px4 documentation](https://docs.px4.io/master/en/peripherals/camera.html) this is also the command sent when the "Camera Trigger" test button is used. I also tried several other commands to trigger a single image, but none worked.
When sending this comand via a companion computer the camera was triggered. So the command seems to be fine.
I found another workaround for the mission. I used MissionPlanner to generate a mission using the same commands and then uploaded it to the flight controller. Everything worked fine. When downloading the mission on qgroundcontrol I noticed that parameter five of the trigger command was set to 107 instead of 1. When saving this plan I noticed that parameter five was set to precisely 106.5353216 for some reason. Using this as a parameter value the camera triggering worked during the mission. I could not find another value for parameter five to make it work. I do not think that this is intentional. Furthemore since everything works using MissionPlanner and manually sending the command from a companion computer with parameter five set to 1, I believe this is a bug in QGroundControl. I just wonder why a 1 as parameter does not work, but using 106.5353216 works. I cannot really think of reasons why QGroundControl should not write the parameter directly to the flight controllers mission.
I also verified that this problem does not only occur in hitl. It also occurs during real flights.

## Log Files and Screenshots
Not sure if useful, but here are some logs:
- QGC console log of uploading and executing the not working plan (Parameter 1):  [not_working_plan.txt](https://github.com/mavlink/qgroundcontrol/files/5759230/not_working_plan.txt)
- Autopilot log from executing the not working plan: [uploaded to flight review](https://review.px4.io/plot_app?log=19e46979-d6f8-4317-80b1-13f3b1901ac9)

- QGC console log of uploading and executing the working plan (Parameter 106.5353216): [working_plan.txt](https://github.com/mavlink/qgroundcontrol/files/5759231/working_plan.txt)

- Autopilot log from executing the working plan: [uploaded to flight review](https://review.px4.io/plot_app?log=65c04da7-2350-4f89-94c3-8811405b50d3)

DonLakeFlyer, 2021-01-04:
This isn't a QGC problem. You are going to need to report a firmware issue for this. Setting param5=1 should work.

michael-fxj, 2021-02-21:
@DonLakeFlyer I mentioned in my issues the reasons why I believe that this is not a firmware bug.
I tried the very same thing with ardupilot and the image trigger still does not work with QGroundControl. So this is certainly not a Firmware issue. It was even tried on several different PCs including Windows and Linux operating system.

DonLakeFlyer, 2021-02-21:
There are a number of things going on here:
* The Param5/Command field of `MAV_CMD_DO_DIGICAM_CONTROL` is set to zero decimal places. So it expects the values to be integers. There is a bug in QGC where if you enter a real number in a field which specifies no decimal places it gets confused. I believe the fact that this field should be an integer is correct. I'm reopening this issue to fix that.
* Using `MAV_CMD_DO_DIGICAM_CONFIGURE` in a mission and using it in a manual `COMMAND_LONG` are two completely different code paths. Just because one works doesn't mean the other will.
* In MP, 106.5353216 is not a camera command. It is that Latitude of the WAYPOINT command that you changed to the camera command.
* I'm not sure if PX4 firmware still supports `MAV_CMD_DO_DIGICAM_CONTROL` in missions. They have mostly moved to the new mavlink camera commands. The new command is `MAV_CMD_IMAGE_START_CAPTURE`. 
* You should just able able to use the camera support build right into the QGC UI which does the right thing with PX4:
[image]

DonLakeFlyer, 2021-02-21:
> * I'm reopening this issue to fix that.

Created separate issue instead: #9468

closed 2021-02-21 by DonLakeFlyer