mavlink/qgroundcontrol #9306: Facing Issues With FTP

kshitizgoyal5, 2021-01-06:
hey i am trying to ftp a file from my drone to gcs using FTPManager class in QGC-master. I am facing the following problem.

[image]

In the logs it can be seen that, we are getting errno5 which indicates all available sessions are already in use. Can anyone Please tell the cause of this error. Thanks in advance.

[image]

DonLakeFlyer, 2021-01-06:
In looking at the FTPManager code I believe there is a bug where sessions are not being reset if the FTP download fails or errors out. The sessions are only reset if download succeeds. That could cause sessions to leak on the vehicle side.

kshitizgoyal5, 2021-01-07:
Thanks for your reply @DonLakeFlyer , I have a doubt that if the session is already assigned for reading so why it is giving errno5 that from what i believe is NoSessionsAvailable. I printed the session id that i am receiving as in AckNak and it is printing 0. In the docs it shows the it has range from 0-255. Why is it giving this particular error cause I am Trying to read the file once.

DonLakeFlyer, 2021-01-07:
Ah sorry. I thought you were getting the Nak from the Open File. It's in the middle of burst sequence! That's just completely wrong on the firmware side. That error makes no sense in the middle of a burst sequence. You should file a bug against ArduPilot. 

> In looking at the FTPManager code I believe there is a bug where sessions are not being reset if the FTP download fails or errors out. The sessions are only reset if download succeeds. That could cause sessions to leak on the vehicle side.

I will create a separate issues for this which is a real QGC problem. But just not what you are hitting.

LorenzMeier, 2021-01-22:
Closing as stale.

closed 2021-01-22 by LorenzMeier