mavlink/qgroundcontrol #9312: External battery source

Igor-Misic, 2021-01-07:
There is an option in the **PX4** configuration **BAT1_SOURCE** that could be set to **External**. 

Documentation for this config (https://dev.px4.io/master/en/advanced/parameter_reference.html)

>  If the value is set to 'External' then the system expects to receive **mavlink battery status messages**. 

Currently, if this option is select this error appears:
[image]

Looking for a proper solution to use an external battery source config with QGC.

DonLakeFlyer, 2021-01-08:
Questions for PX4 firmware folks:
* Should `BAT#_SOURCE` parameter setup be added to the Power component ui?
* If so, what does that mean with respect to it being set to "External" or "ESCs"?
* Does it mean for those the only thing shown is `BAT#_SOURCE` combo and then no other UI is needed?

DonLakeFlyer, 2021-01-17:
> * Should `BAT#_SOURCE` parameter setup be added to the Power component ui?

It's already there. It was added to QGC mainly to be able to disable batteries.

DonLakeFlyer, 2021-01-17:
If think if external it means most of these settings are not there.

Igor-Misic, 2021-01-31:
@DonLakeFlyer I wanted to close this issue because when I run the latest master PX4,  this error doesn't appear, but now I have another one:
QGC v4.1.1
[image]

DonLakeFlyer, 2021-01-31:
> There is an option in the **PX4** configuration **BAT1_SOURCE** that can be set to **External**.

This problem hasn't been fixed yet and still exists in QGC. If you still have that setup you are still going to see problems. They can be ignored through since there is no additional setup needed for External.

Igor-Misic, 2021-01-31:
So this is a known issue. 
Let me know if there is any PR so we can close this. Thanks!

closed 2023-06-05 by Igor-Misic