mavlink/qgroundcontrol #9314: updateqrc.py code python version

mnumanuyar, 2021-01-08:
fallowing line throws error in python 3, it is an easy fix, and error mesage itself show how to fix it and there are online converters avaliable.
Should ve change it, or should we mention it anywhere ?
Error :
```
 File "updateqrc.py", line 28
    print 'Excluded:', line
          ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print('Excluded:', line)?
```
line:
https://github.com/mavlink/qgroundcontrol/blob/44be3032b1f49ab0ea2dc540b522746eed7bf1b2/custom-example/updateqrc.py#L28
fix : 
`  print('Excluded:', line)`

DonLakeFlyer, 2021-01-08:
It should be changed. Can you submit a pull? I just fixed the same exact thing in the localization python scripts.

DonLakeFlyer, 2021-01-10:
I was already in the custom code for other reason so I'm fixing this myself.

closed 2021-01-10 by DonLakeFlyer

mnumanuyar, 2021-01-11:
I just saw your response, thank you for fixing it

PR #9323 (2021-01-10): Custom example update
Fix #9314, #9186