mavlink/qgroundcontrol #9319: Implement better GStreamer debugging with respect to which plugin was chosen

DonLakeFlyer, 2021-01-08:
So we get a lot of reports of "my video doesn't work in QGC". To me it should be 100 times easier to debug these sorts of things remotely. Without that capability we will spend the rest of our lives going back and forth with QGC users trying to figure out why with limited information available to go on.

Problems I see:
* There are some video streams we support and some we specifically do not support. If someone tries to play from a video stream we don't support QGC should specifically pop an error saying we don't support that and why. As it currently stands the way this is currently handled is that you just don't get video to be played.
* There are some video streams we support and within those supported streams there seem to be only some specific variants/options we support. Same as above, if the video stream is not the supported variant or caps or whatever gstreamer want the end result is no video plays. No helpful information to the user as to why.
* The gstreamer pipeline decides from parsebin and decodebin which plugin to use. Some of these plugins are good some are bad. Sometimes the creator of vehicles which support video want to understand what actual plugins get used for their video stream. As far as I can tell there is not sane way to currently understand what plugin(s) are actually being used.

As far as I can tell the only way to debug video not working problems is to turn on a gst debug level and then wade through megabytes of gst debug output to maybe fine some possible needle in haystack. It all seems like insanity to me. There has to be a better way to do this. There is no way to support video in QGC as it currently stands.

patrickelectric, 2021-01-28:
> * There are some video streams we support and some we specifically do not support. If someone tries to play from a video stream we don't support QGC should specifically pop an error saying we don't [...]

That's totally possible and we should indeed end up doing that.

> * There are some video streams we support and within those supported streams there seem to be only some specific variants/options we support. Same as above, if the video stream is not the supported [...]

That's also totally possible and related to the first point.

 

> * The gstreamer pipeline decides from parsebin and decodebin which plugin to use. Some of these plugins are good some are bad. Sometimes the creator of vehicles which support video want to understand what actual plugins get used for their video stream. As far as I can tell there is no sane way to currently understand what plugin(s) are actually being used.

We can say which plugins are being used, what we can't say which plugins _**will**_ be used, because the usage is defined by gstreamer algorithm for the best possible scenario, and that can depends of gstreamer version, video drivers installed on the host machine, cpu type (if supports video decode), operating system codec and video card if there is one.

I believe that is a feature that must be implemented.

patrickelectric, 2021-01-28:
PS: It would be nice to have a "must-have", "good-to-have" and "nice-to-have" tags for priority and importance.

DonLakeFlyer, 2021-01-28:
> We can say which plugins are being used, what we can't say is which plugins _**will**_ be used

That's fine. The important thing is to be able to log what was actually used.

farhangnaderi, 2021-02-04:
@Mustafadagkiranlar