mavlink/qgroundcontrol #9331: qground 4.1.0 linux sitl connection, cant send message to vehicle

mnumanuyar, 2021-01-14:
## Expected Behavior
After SITL connection gcs should be able to read parameters and write mission.

## Current Behavior
I gt warnings like 
"Vehicle 1 did not respond to request for parameters. This will cause QGroundControl Daily to be unable to display its full user interface"
"Mission transfer failed. Error: Mission write mission count failed, maximum retries exceeded."

but I am able to get vehicles data. For example if vehicle is moving, I can see it move in qgc.

## Steps to Reproduce:
1.start sitl
2.start qground
3.do anything that sends message from gcs to vehicle.

## System Information
When posting bug reports, include the following information
- kubuntu 18.04
- QGC Version: 4.1.0
- QGC build: stable, self build. also daily build from [here](https://docs.qgroundcontrol.com/master/en/releases/daily_builds.html) which also says 4.1.0
- Flight Controller: SITL
- Autopilot (with version): latest (commit sha: d02a61a65da06437172127e14d33de3da0eb08e0)

## Detailed Description
I have no problem with communicating with a real veicle. Also I can work with an early stable build of QGC with my current setup. 
The problem is happening with only 4.1.0 and SITL. (I have tried two versions of sıtl)

## Log Files and Screenshots

[qgcSITL.txt](https://github.com/mavlink/qgroundcontrol/files/5814850/qgcSITL.txt)

mnumanuyar, 2021-01-21:
additionaly, I observed that, QGC 4.1.0 did not get forwarded messages from sitl. where QGC 4.0.11 shows them.
Details  : 
after I connect sitl to gcs on port 5760, i am connecting sitl port 5763 with a separate python script, and simulate a component connected to autopilot. from here I send status text messages targeted to gcs. I did not saw these messages in mavlink inspector but, in v4.0.11 , I am able to see them

DonLakeFlyer, 2021-01-21:
I think there is a problem with TCP connections, trying to repro

DonLakeFlyer, 2021-01-22:
@mnumanuyar Can you try this fix: https://qgroundcontrol.s3-us-west-2.amazonaws.com/builds/Stable_V4.1/QGroundControl-installer.exe

mnumanuyar, 2021-01-22:
hello, 
I built commit ceff9093cef8c9abb8ccb77a7ef5092c342e2694 (current head of Stable_V4.1 ) and tried in Linux, It works perfectly, 
I dont have windows with SITL with me right now, I will be able try the install file monday.
thank you very much.

DonLakeFlyer, 2021-01-22:
Ok, perfect that's all I need

closed 2021-01-22 by DonLakeFlyer

wetouns, 2021-02-01:
> @mnumanuyar Can you try this fix: https://qgroundcontrol.s3-us-west-2.amazonaws.com/builds/Stable_V4.1/QGroundControl-installer.exe

has android installer?