mavlink/qgroundcontrol #9334: Stable v4.1 release problems

DonLakeFlyer, 2021-01-14:
* Windows: Seems to be installing a daily build or possibly tagged builds are broken
* OSX: No app name in title bar. About shows correct versioning.

AlexKlimaj, 2021-01-15:
I am seeing this same issue on windows.

AlexKlimaj, 2021-01-15:
Also, when I connect to a com link, disconnect is not working.

DonLakeFlyer, 2021-01-18:
Android 4.1 is available on play store now

Radanz, 2021-01-18:
Same issue here it seems: Windows X64 v4.1 downloaded from website (Amazon bucket) and from github releases
Title bar shows "Daily"
Connects to flight controller but wont allow setup. Shows Telemetry ok and I can flash the firmware, just nothing fills in the setup screen.
Github release said it was only an hour old so must be a fresh update? or the daily is pushing to the stable for some reason?
Rolling back to previous version for now.

DonLakeFlyer, 2021-01-18:
@Radanz Can you try the same thing with a Daily build and if still broken create a new issue with full details.

DonLakeFlyer, 2021-01-19:
Windows build now correctly marked as Stable

closed 2021-01-22 by DonLakeFlyer