mavlink/qgroundcontrol #9336: App crash when using Follow Terrain (repeatable)

Antiheavy, 2021-01-15:
Steps to recreate application crash: 
(Win10, daily build)

1) launch QGC
2) go to Plan screen
3) place Takeoff
4) place a Survey
5) click mountains icon
6) check box for follows terrain
7) enter a value for max decent rate
8) app crashes

Now I can't get into the screen to change the value back to 0 because it insta crashes whenever I check that box on again!

Antiheavy, 2021-01-15:
Got any hints on what file i need to nuke so that QGC doesn't remember that Max Descent Rate value?

DonLakeFlyer, 2021-01-15:
Just clear your settings. Available from Misc section of App Settings

Antiheavy, 2021-01-15:
> Just clear your settings. Available from Misc section of App Settings

oh yeah!  thanks.   FWIW this bug has been around for at least a few weeks as I was testing an older daily when i first stumbled into it.

closed 2021-01-22 by DonLakeFlyer