mavlink/qgroundcontrol #9349: Linux application opens in an un-exitable, full screen mode

djcopley, 2021-01-19:
## Expected Behavior
I expect the ability to put the Ground Control software into a windowed mode.

## Current Behavior
When I launch the app image, it starts in a full-screen mode that appears to be un-exitable.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Double click QGroundControl.appimage

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 20.04
- QGC Version: v4.1.0
- QGC build: stable
- Flight Controller: Pixhawk
- Autopilot (with version): Irrelevant

## Detailed Description
QGroundControl starts in a fullscreen window that cannot be closed.

## Log Files and Screenshots
[image]

DonLakeFlyer, 2021-01-21:
Wha is your screen resolution?

djcopley, 2021-01-21:
> What is your screen resolution?

5120x1440

djcopley, 2021-02-09:
I have found the problematic qml causing this bug.

https://github.com/mavlink/qgroundcontrol/blob/36f7c2212f1ea378a0629ca8fe43f8bb06356301/src/ui/MainRootWindow.qml#L33

The check `Screen.height / ScreenTools.realPixelDensity < 120` returns true for 1440p super ultrawide monitors.

EnsignCrasher, 2021-02-26:
i had a similar problem and solved it by changing
```
[MainWindowState]
visibility=5
```
to
```
[MainWindowState]
visibility=2
```
in ~/.config/QGroundControl.org/QGroundControl.ini

DonLakeFlyer, 2021-02-26:
@EnsignCrasher Are you running at a really low res screen or small screen in actual dimensions?

closed 2021-02-26 by DonLakeFlyer

EnsignCrasher, 2021-02-26:
gpd micropc 6" 1280x720^^
i'm not quite sure how this happend though, imo it just started in fullscreen some day... But I don't want to rule out the possibility that I made some mistake...^^

deonblaauw, 2022-05-24:
I'm experiencing this issue with QGC v4.2.1 on Ubuntu 20.04 LTS. Any tips on how to work around this?

AllenHuangDongyue, 2023-10-14:
> i had a similar problem and solved it by changing
> 
> ```
> [MainWindowState]
> visibility=5
> ```
> 
> to
> 
> ```
> [MainWindowState]
> visibility=2
> ```
> 
> in ~/.config/QGroundControl.org/QGroundControl.ini

Doesn't help under this solution. Parameter is changed back automatically after running QGC. Problem still exists.