mavlink/qgroundcontrol #9357: The latest version（4.1.0） of the software cannot be set to Chinese

liu-jiaming, 2021-01-20:
The latest version（4.1.0） of the software cannot be set to Chinese.
The language is set to follow the system, set to Chinese, and the English interface is still displayed.
This version is easy to flash back. It's easy to flash back when you use it.
The computer system is win10 Enterprise Edition.
The following log was flashed back once during the logging process.
[5353.txt](https://github.com/mavlink/qgroundcontrol/files/5841988/5353.txt)

DonLakeFlyer, 2021-01-22:
Looks like the major of Chinese translations are not yet approved

xdwgood, 2021-03-04:
What can be done to solve this problem?

DonLakeFlyer, 2021-03-04:
Both the Chinese and Korean translation managers need to approve the strings in Crowdin.

kaiy860123, 2021-07-06:
The Chinese translation managers have no approving activity since Nov 09, 2020. Maybe assign another manager?

booo, 2021-10-10:
Similar issue here https://github.com/mavlink/qgroundcontrol/issues/9439