mavlink/qgroundcontrol #9381: Unable to send command: Waiting on previous response to same command.

lovettchris, 2021-01-28:
## Expected Behavior
Should be able to callibrate the radio with out any problems.

## Current Behavior
Calibrating radio fails to send final command to Pixhawk 4 when calibration is completed.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Plugin in Pixhawk 4 with firmware version 1.11.2, connect a Futaba T14SG radio control to the Pixhawk
2. Download and install QGroundControl version 4.1.1 from http://qgroundcontrol.com/downloads/
3. Launch QGroundControl on Windows and wait for it to find the Pixhawk serial port.
4. Run radio control calibration and click "Send" to send calibration to the pixhawk.

## System Information
When posting bug reports, include the following information
- Operating System: Windows 10
- QGC Version: 4.1.1
- QGC build: stable
- Flight Controller: Pixhawk 4
- Autopilot (with version): PX4 1.11.2

## Detailed Description
No matter what I try I keep getting this error: "Unable to send command: Waiting on previous response to same command."

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
[qgroundcontrol.txt](https://github.com/mavlink/qgroundcontrol/files/5890616/qgroundcontrol.txt)

- Autopilot logs when available (post a link).
While tracing mavlink messages sent from QGroundControl using "Enable Mavlink Forwarding" I found only these and nothing about calibration, so looks like it never sent any message ?

```json
{ "rows": [
    { "name": "COMMAND_LONG", "id": 76, "timestamp":0, "msg": {"param1":0, "param2":0, "param3":0, "param4":0, "param5":0, "param6":0, "param7":0, "command":520, "target_system":255, "target_component":0, "confirmation":1} },
    { "name": "PARAM_REQUEST_LIST", "id": 21, "timestamp":0, "msg": {"target_system":1, "target_component":1} },
    { "name": "COMMAND_LONG", "id": 76, "timestamp":0, "msg": {"param1":0, "param2":0, "param3":0, "param4":0, "param5":0, "param6":0, "param7":0, "command":520, "target_system":1, "target_component":1, "confirmation":1} },
    { "name": "COMMAND_LONG", "id": 76, "timestamp":0, "msg": {"param1":144, "param2":0, "param3":0, "param4":0, "param5":0, "param6":0, "param7":0, "command":176, "target_system":1, "target_component":1, "confirmation":1} },
    {}
]}

```

- Screenshots of QGC to help identify the current issue/bug behavior.
[image]

DonLakeFlyer, 2021-01-29:
Can you turn `VehicleLog` logging and capture a console log?

lovettchris, 2021-01-29:
Sorry can you be more specific, I'm not finding the term "VehicleLog" anywhere in https://docs.qgroundcontrol.com/master/en/SettingsView/General.html search index.  Note: I did paste a "Console log" output in the above - see the attached "qgroundcontrol.txt" file.

DonLakeFlyer, 2021-01-29:
https://docs.qgroundcontrol.com/master/en/SettingsView/console_logging.html

DonLakeFlyer, 2021-01-29:
FYI: If you ignore the error nothing bad will happen

lovettchris, 2021-01-29:
Here's the "VehicleLog" captured by console log:

[consolelog.txt](https://github.com/mavlink/qgroundcontrol/files/5896904/consolelog.txt)

the error is confusing because it makes one believe the calibration failed.

DonLakeFlyer, 2021-01-30:
The fix for this is already in daily builds. When I decided to mark stable 4.1.2 as next release this will be in stable as well.

lovettchris, 2021-01-31:
Cool !

kadirozgunn, 2021-02-10:
Hi, 
I have same problem with Mac OS Big Sur 11.2 and QgroundControl v4.1.1 I could not see daily build 4.1.2 for Mac. I got the problem when I connect the Joystick. And also when I plug out joystick QGroundControl crash. Can you offer any solution for now?

DonLakeFlyer, 2021-02-16:
https://docs.qgroundcontrol.com/master/en/releases/daily_builds.html

closed 2021-02-16 by DonLakeFlyer

SemvdH, 2023-03-31:
Hello, I also have this problem with QGC v4.1.7 on Ubuntu 22.04 and PX4 v 1.13. I got the same error when trying the motor test through the QGC `motors` settings. When I tried to use the MAVLink Console however, I could control each motor individually and also test them all at the same time, without error. I did this with the command `motor_test -p 50 -t 3` to test all motors at 50% power for 3 seconds.