mavlink/qgroundcontrol #9388: QGC mavlink on INAV using TBS Crossfire (wifi connection)

rhys100, 2021-02-02:
I spent many hours trying to get mavlink working on inav so i could see where my plane or quad was which was running INAV software and crossfire. As Crossfire now supports a wifi connection, theoretically you can connect your ipad/android tablet to the crossfire wifi AP and then hit connect on QGC to see your craft using mavlink protocol.

However it seems this does not work. What would be needed to make QGC work with it again?

From my research here is my findings

INAV only using mavlink telementry, NOT MSP.
"The wiki has been updated to reflect the claim that contemporary qgroundcontrol no longer processes iNav mavlinbk telemetry (not wihstanding the fact that qgroundcontrol's telemetry log records valid Mavlink data (which is processed as valid mavlink telemetry by mwp)"
https://github.com/iNavFlight/inav/issues/5059

Further topics on this discussion that have no answers
https://github.com/iNavFlight/inav/issues/6164
https://github.com/mavlink/qgroundcontrol/issues/6616
https://github.com/mavlink/qgroundcontrol/issues/4857
https://www.facebook.com/groups/1309125302467027

More info on tbs crossfire wifi
https://team-blacksheep.freshdesk.com/support/solutions/articles/4000161703-tbs-crossfire-update-the-wifi-module-
https://www.team-blacksheep.com/tbs-crossfire-mavlink-over-wifi.pdf

LorenzMeier, 2021-02-02:
Are you able to build QGC from source? I could point you to the locations to instrument.

rhys100, 2021-02-02:
Sorry thats way over my head

vidma, 2021-03-06:
I think i have the same issue with Ardupilot. 
I'm a developer so I should be able to build QGC (preferably assuming easy tooling setup exists for OSX and it wouldn't take much time). @LorenzMeier 

I have crossfire diversity nano RX + Opentx radiomaster tx16s with `TBS Micro WiFi transmitter`.
The issue  is that **QGC never gets connected to the Mavlink proxy (UDP) which should be created by TBS WiFi. I think I tried both mobile phone & laptop.**

According to docs  it seems enabling only CSRF should be enough to get Mavlink. There's also other option of connecting both CSRF + Mavlink RX/TX together, but CSRF seems to be a more performant protocol for RX to TX communication (my guess).

vidma, 2021-03-06:
btw, interestingly, when using UDP over WiFi if i select `high latency` when adding a connection in QGC (not sure if required) & look only at Analyze tools -> Mavlink inspector, I see some Radio status "messages". 

[image]

but QGC still says disconnected! 
[image]

vidma, 2021-03-06:
another sad thing is that this mavlink over Wifi  if enabled seem to slow down (or make even stuck) the `yaapu telemetry widget` on the RC transmitter/controller (or at least it looks so).

psb-24cq, 2021-07-02:
@vidma, Has this issue been resolved, and have you been able to connect the Crossfire telemetry through wifi to QGC? I am attempting to do the same thing as mentioned in this post and cannot get it to work.

rhys100, 2021-07-02:
@psb-24cq i believe its still an issue

esterlinpol, 2021-07-11:
In my case I tried to use mavlink on separate UARTS to output telemetry to the zs6buj Antenna Tracker but when the connection is active, the telemetry and GPS Packages have an increased loss, 

Any advise for this? 

I have the Micro tx on a radiomaster tx16s, and the crossfire nano rx on the plane 

Is there a way to transfer LTM instead over crossfire then output it over WiFi?

vidma, 2021-07-12:
I tried RaspberryPi Zero (on airplane) with https://www.docs.rpanion.com/software/rpanion-server over 4G or WiFi
when telemetry is sent *over UDP* connection to QGC (in rpanion UDP output settings enter the IP address of QGC, with default udp port of QGC - dont remember which it is -> google up), it works pretty well even including video streaming or recording (I use my own code hack/patch for recording)...

steinne, 2022-02-01:
Is this issue with TBS wifi and lost mavlink packets still active? I am having similar issues.

vidma, 2022-02-01:
no idea if helps (didn't try wifi mavlink anymore again), but TBS mentioned had recent related improvements - upgrade your TBS crossfire firmware on your RX/TX modules (via TBS Agent)

steinne, 2022-02-01:
> no idea if helps (didn't try wifi mavlink anymore again), but TBS mentioned had recent related improvements - upgrade your TBS crossfire firmware on your RX/TX modules (via TBS Agent)

Appreciate it-   updated all items to latest. (WIFI/Crsf) still seeing significant packet loss.  With the new RX Pro (500mw telemetry) was hoping I would have a simple longer range Groundstation radio solution.    May TBS will update the WIFI soon to be more robust,    Thanks for your feedback

esterlinpol, 2022-02-01:
Hello, 

As I couldn't make this work, I had to look for an alternative, 

What I ended up doing is that I purchased two hc-12 wireless serial modules at Amazon for the telemetry. They are really cheap and can give you about 2 km or more of range if you use a bigger antenna.

steinne, 2022-02-01:
> Hello,
> 
> As I couldn't make this work, I had to look for an alternative,
> 
> What I ended up doing is that I purchased two hc-12 wireless serial modules at Amazon for the telemetry. They are really cheap and can give you about 2 km or more of range if you use a bigger antenna.

cool....  would you be able to share the link?      (would have been a monster win if TBS was able to pull this off)

vidma, 2022-02-02:
> What I ended up doing is that I purchased two hc-12 wireless serial modules at Amazon for the telemetry. They are really cheap and can give you about 2 km or more of range if you use a bigger antenna.

and if you use `wifibroadcast` or https://github.com/OpenHD/Open.HD with very big receiver antenna you could in theory get up to 80+km range, including HD video.

pedrodparkes, 2022-09-03:
no success even now, 2y after the thread creation

MarioFPVdev, 2022-12-31:
> > What I ended up doing is that I purchased two hc-12 wireless serial modules at Amazon for the telemetry. They are really cheap and can give you about 2 km or more of range if you use a bigger antenna.
> 
> and alternatively if you use `wifibroadcast` or https://github.com/OpenHD/Open.HD with very big ground antenna you could in theory get up to 80+km range, including HD video (these are raspberry pi based)

QOpenHD has at least 2x video latency compared to QGroundControl so it's not ideal for FPV flights.

hcolsen, 2023-01-04:
I hope a solution is nearby.. Program looks awesome! I will try it tomorrow
TBS Tango 2, xf nano rx, wifi, mavlink rx/tx on separate uarts on diatone mamba f722App mk4 with mateksys m8q w. mag running inav 5.1

Niko-La, 2023-11-27:
> QOpenHD has at least 2x video latency compared to QGroundControl so it's not ideal for FPV flights.

is this still true for 2023 Dec?

trying to get INAV fc to talk to Linux GS looking for one not sure which is the right one QOpenHD or QGroundStation.  RC is via ELRS but want to establish a separate data one with an android smartphone

closed 2026-10-04 by github-actions[bot]