mavlink/qgroundcontrol #9392: Fonts in iOS 14.4 are rendered wrong

edupaz2, 2021-02-03:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
Fonts are properly displayed.

## Current Behavior
Spacing and rendering of the fonts are wrong. Some characters are cut in half.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Download latest version in AppStore (4.0.10).
2. Open the app.
3.
4.

## System Information
When posting bug reports, include the following information
- Operating System: iOS 14.4
- QGC Version: 4.0.10

## Detailed Description
With iOS 14.3 the app fonts looked good but with the latest upgrade iOS 14.4 not.

## Log Files and Screenshots
[image]

edupaz2, 2021-02-04:
Apparently with Qt 5.14.2 the issue is gone.

Antiheavy, 2021-02-07:
ah ha!   I saw this recently too on the previous version of QGC build for iOS.   FYI @LorenzMeier 

> Apparently with Qt 5.14.2 the issue is gone.

does the newest QGC use the newer version of Qt?

edupaz2, 2021-02-08:
@Antiheavy I haven't checked that.

taileron, 2021-02-11:
The iOS update to 14.4 also affects all older QGC, so 3.2 and 3.4.3 are also no longer usable right now and I am dependent on the old versions because of realtime rtsp. I can also remember that the same behaviour occurred a long time ago in the same form there was also a solution without changes to QGC ... I just don't remember what it was

philou91, 2021-02-15:
I would be interested as well, since I couldn't prevent my two IPhones to upgrade to iOS 14.4 and QGC is almost unusable now. By the way, can you explain how often iOS releases  are pushed to app-store ? The current one is 6 months old (4.0.10)

dakejahl, 2021-04-06:
Any movement on this? Any plan for the near term?

>Apparently with Qt 5.14.2 the issue is gone.

Was this something you tested @edupaz2 ?

edupaz2, 2021-04-06:
@dakejahl I didn't test with QGroundControl itself but with another app. I can confirm the issue is gone, although the fonts don't look as neat as before, but not that bad.