mavlink/qgroundcontrol #9394: Virtual joystick: not sending when window inactive

coderkalyan, 2021-02-04:
## Expected Behavior
When the QGroundControl app is running but is minimized, it stops sending manual control setpoints from the on screen virtual joystick, and PX4 starts spamming errors as a result "manual control lost."

## Current Behavior
QGC should continue sending the same setpoint as previously set when the window is minimized.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Enable virtual joystick in QGC settings
2. Connect to FC over USB (SITL should work too).
3. Minimize the window
4. Observe 

## System Information
When posting bug reports, include the following information
- Arch Linux, Wayland
- QGC Version: 4.0.11
- QGC build: stable (AUR)
- Flight Controller: CUAV V5+
- Autopilot (with version): PX4 1.11.2 stable
- 
## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.
[image]
Autopilot log:
[image]

DonLakeFlyer, 2021-02-16:
Do you have some sort of settings turned on for your OS which caused an App to stop running when it is minimized? I just tested this on Ubuntu 20.04 and it is working fine for me.

coderkalyan, 2021-03-01:
Let me do some digging, good to know that it's just me :smiley: