mavlink/qgroundcontrol #9396: Menu appears over hamburger button in 4.1.1

julianoes, 2021-02-04:
## Expected Behavior
I want to click the hamburger button to have the side menu appear and disappear.

## Current Behavior
The menu appears on top of the button.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Press hamburger menu

## System Information
When posting bug reports, include the following information
- Operating System: Arch LInux
- QGC Version: 4.1.1
- Flight Controller: SITL
- Autopilot (with version): PX4 master

## Log Files and Screenshots

[image]

Antiheavy, 2021-02-07:
I suspect if the list of menu items was shifted down one space it would have the effect you are looking for, although without the "hamburger" icon.
[image]

julianoes, 2021-02-08:
> Are you looking for a "toggle" type function on that button?

Yes, and that's what I remember what it always was.

closed 2021-02-21 by DonLakeFlyer