mavlink/qgroundcontrol #9398: Can't plan a mission without takeoff

julianoes, 2021-02-04:
## Expected Behavior
I used to be able to create a mission without having to care about a takeoff item. Having to add a takeoff item if you are already in the air does not make sense in my opinion.

## Current Behavior
With 4.1.1 it no longer allows me to create a waypoint mission without clicking takeoff first.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Takeoff
2. Go to plan view
3. Try to create a mission (waypoint is greyed out)

## System Information
When posting bug reports, include the following information
- Operating System: Arch Linux
- QGC Version: 4.1.1
- QGC build: stable
- Flight Controller: SITL
- Autopilot (with version): PX4 master

Antiheavy, 2021-02-07:
Questions:
- should there be special handling when there is a live connection to a vehicle and the vehicle is in the air?
- should there be a General Settings checkbox option to "require a Takeoff first" in every mission or to allow the user to decide?  
 
I suppose there are some users that never want to use a Takeoff waypoint as they always takeoff via manual or maybe a special takeoff flight "mode".

We operate a fixed wing and the only way for it to properly takeoff is by using a Takeoff waypoint.  i can say from lots of experience that our users are not smart enough to decide for themselves when they need a Takeoff and when it is okay to not have a Takeoff. The fact that QGC requires them to plan a Takeoff before anything else is a huge safety benefit for us.

DonLakeFlyer, 2021-02-07:
You can turn off the requirement for takeoff from App Settings - Plan View section

DonLakeFlyer, 2021-02-07:
Thinking is that Missions which start in the air aren’t common enough for require takeoff to not be default.

Antiheavy, 2021-02-07:
> You can turn off the requirement for takeoff from App Settings - Plan View section

So you can!  thanks.   screenshot:
[image]

julianoes, 2021-02-08:
I see, thanks!

closed 2021-02-08 by julianoes

OmerPorat, 2023-08-01:
hey @julianoes!
I encounter the same issue when uploading a mission to an airborne drone. Even though I did not upload a takeoff command the drone starts ascending and only then goes to the first waypoint. Is there a way to turn off the requirement for takeoff on the PX4 side or on the Mavlink commands I am sending?
Thanks!

julianoes, 2023-08-07:
https://github.com/mavlink/qgroundcontrol/issues/9398#issuecomment-774770582 ?