mavlink/qgroundcontrol #9399: Flight mode button crammed between symbols

julianoes, 2021-02-04:
## Expected Behavior
The top bar used to be somewhat nicely laid.

## Current Behavior
With 4.1.1 suddenly everything is crammed to the left. Especially the flight mode is crammed in between menu and symbols.

This can even get dangerous because double clicking (with some noise) on the flight mode button will switch to manual (see screenshot below).

## System Information
When posting bug reports, include the following information
- Operating System: Arch Linux
- QGC Version: 4.1.1
- QGC build: stable
- Flight Controller: SITL
- Autopilot (with version): PX4 master

## Log Files and Screenshots

[image]

Antiheavy, 2021-02-07:
Yes, the current layout is too crowded. 

> This can even get dangerous because double clicking (with some noise) on the flight mode button will switch to manual (see screenshot below).

oh no!

Antiheavy, 2021-02-07:
some thoughts for what it's worth (sorry for the terrible "photoshop"):
[image]

DonLakeFlyer, 2021-02-07:
I’ll add some comments here on reasoning for current setup when I get back in town. Double click thing is certainly badness.

DonLakeFlyer, 2021-02-18:
For reference visually:

4.0
[image]

4.1 on the ground
[image]

4.1 in the air
[image]

My opinion:
* I certainly agree that the double-click thing is a nasty problem that needs fixing. The menu should position itself below the toolbar when it opens to prevent possible double-click hits.
* In general I find the new layout way cleaner and significantly less "crammed"  or "noisy" than the previous layout in 4.0
* The margin on the left/right of the flight mode text are unequal. This causes things to look uneven and not spaced well. 
* The left/right margin around the flight mode label is also a bit too small in that it doesn't create enough separation from the other elements.
* The flight mode icon has a pixel on the right clipped which makes it look odd

Here is what it looks like with the things listed above changed (in air example):
[image]
To me this is significantly better than 4.0

julianoes, 2021-02-18:
If you space it out a bit more it would be better. The problem I see is that with the new design it's not clear that the mode text is also a button to change the mode. So that's worse UX wise, in my opinion.

Antiheavy, 2021-02-18:
> If you space it out a bit more it would be better. The problem I see is that with the new design it's not clear that the mode text is also a button to change the mode. So that's worse UX wise, in my opinion.

I 100% agree.   The status message ("Flying") and the squiggle, and the flight mode all need more space around them for 2 reasons: 1) to be less confusing and 2) it's too prone to fat fingers hitting the wrong spot.

> To me this is significantly better than 4.0

Yes!  the new toolbar is heading in the right direction!  just needs a little tweaking to space things out a bit.  IMHO.

DonLakeFlyer, 2021-02-18:
[image]

DonLakeFlyer, 2021-02-18:
Ok, with this last set of changes I think it's good for Stable. It could certainly use more work but I'd rather let it sit and gather more feedback ground users and then come back to it in master.

DonLakeFlyer, 2021-02-18:
> The problem I see is that with the new design it's not clear that the mode text is also a button to change the mode. So that's worse UX wise, in my opinion.

Hmm, I get that. Nothing in the toolbar looks like a button buts it all clickable. I wonder whether that will be a problem that people figure out pretty quickly. We'll see. If it ends up being totally confusing I can rework.

julianoes, 2021-02-19:
That's certainly safer, thanks @DonLakeFlyer.

julianoes, 2021-03-04:
I just tested this and while I think it's better I find it suboptimal that it doesn't toggle on/off like all other modals that appear/disappear when you click the top bar. 

Right now, to make it disappear you need to click somewhere else, just clicking the button in the top bar again, makes it reappear.

booo, 2021-10-15:
@julianoes can you create another ticket for this specific problem? With your nice video attached and a description which behavior we are looking for? Otherwise a (new) developer has to go through the whole ticket and has to identify the peaces that still need work.