mavlink/qgroundcontrol #9406: active vehicle change freezes ui if there is too much values in instrument value widget

mnumanuyar, 2021-02-05:
## Expected Behavior
proceses related to active vehicle change should not lock ui

## Current Behavior
if there is a lot of values in InstrumentValue Widget, changeing active vehicle locks up ui

## Steps to Reproduce: 
1. open gcs
2. connect 2 vehicles
3. go to flyview
4. add alot of value in Instrument value widget at h-the buttom of screen. (I have 4*5)
5. change vehicle using drop down at the top of screen
6. move mouse to plan view button at left side of screen (it wont highlight) click it (plan view will open after a delay)

## System Information 
- Operating System: kubuntu 18.04 
- QGC Version: 4.1
- QGC build: daily. sha: 9589cc5f002d55099954428603413a6a00df20fa
- Flight Controller: sitl
- Autopilot (with version): sha: f0d8ca5e371b7f8b1c1b883fed87043d98d2cfe3 

## Detailed Description
slow down increases as the value count increases in the Instrument value widget, but it is visible even if there is 2*2 values there.

## Log Files and Screenshots
[QGCInstrumentWidgetAlternateBug.txt](https://github.com/mavlink/qgroundcontrol/files/5930995/QGCInstrumentWidgetAlternateBug.txt)

DonLakeFlyer, 2021-02-22:
Can you give me exact instructions to start and connect to multiple ArduPilot SITL vehicles? It's not reproducing for me so far.

DonLakeFlyer, 2021-03-27:
Just tried to reproduce again. All working fine for me.

mnumanuyar, 2021-03-31:
sry I couldnt spare time here, I am still seeing this problem, I could look more into it ~6 weeks later

closed 2026-10-04 by github-actions[bot]