mavlink/qgroundcontrol #9416: UDP h.264 Video Stream not rendered on Windows 10

dorovl, 2021-02-06:
While the UDP h.264 Video Stream is well rendered on Android 32/64 bit, there is a problem on Windows 10.

This has been reproduced in v4.1.0. It works well on v4.0.11.

[image]

tzuriel-spearuav, 2021-02-15:
Seeing the same issue with nextvision camera (works well on 4.0.11)
Console shows following errors :
```
[E] at ../gst/videoparsers/gsth264parse.c:1492 - "<h264parse0> broken/invalid nal Type: 1 Slice, Size: 8820 will be dropped"
[E] at ../gst/videoparsers/gsth264parse.c:1492 - "<h264parse0> broken/invalid nal Type: 1 Slice, Size: 8559 will be dropped"
[E] at ../gst/videoparsers/gsth264parse.c:3450 - "<h264parse1> H.264 AVC caps, but no codec_data"
[E] at ../gst/videoparsers/gsth264parse.c:3476 - "<h264parse1> refused caps video/x-h264, stream-format=(string)avc, alignment=(string)au, width=(int)1280, height=(int)720, framerate=(fraction)0/1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, profile=(string)constrained-baseline, level=(string)3.1"
[E] at ../gst/videoparsers/gsth264parse.c:3450 - "<h264parse1> H.264 AVC caps, but no codec_data"
[E] at ../gst/videoparsers/gsth264parse.c:3476 - "<h264parse1> refused caps video/x-h264, stream-format=(string)avc, alignment=(string)au, width=(int)1280, height=(int)720, framerate=(fraction)0/1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, profile=(string)constrained-baseline, level=(string)3.1"
[E] at ../gst/videoparsers/gsth264parse.c:3450 - "<h264parse1> H.264 AVC caps, but no codec_data"
[E] at ../gst/videoparsers/gsth264parse.c:3476 - "<h264parse1> refused caps video/x-h264, stream-format=(string)avc, alignment=(string)au, width=(int)1280, height=(int)720, framerate=(fraction)0/1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, profile=(string)constrained-baseline, level=(string)3.1"
[E] at ../gst/videoparsers/gsth264parse.c:3450 - "<h264parse1> H.264 AVC caps, but no codec_data"
[E] at ../gst/videoparsers/gsth264parse.c:3476 - "<h264parse1> refused caps video/x-h264, stream-format=(string)avc, alignment=(string)au, width=(int)1280, height=(int)720, framerate=(fraction)0/1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, profile=(string)constrained-baseline, level=(string)3.1"
[E] at ../gst/videoparsers/gsth264parse.c:3450 - "<h264parse1> H.264 AVC caps, but no codec_data"
[E] at ../gst/videoparsers/gsth264parse.c:3476 - "<h264parse1> refused caps video/x-h264, stream-format=(string)avc, alignment=(string)au, width=(int)1280, height=(int)720, framerate=(fraction)0/1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, profile=(string)constrained-baseline, level=(string)3.1"
[E] at ../gst/videoparsers/gsth264parse.c:3450 - "<h264parse1> H.264 AVC caps, but no codec_data"
[E] at ../gst/videoparsers/gsth264parse.c:3476 - "<h264parse1> refused caps video/x-h264, stream-format=(string)avc, alignment=(string)au, width=(int)1280, height=(int)720, framerate=(fraction)0/1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, profile=(string)constrained-baseline, level=(string)3.1"
[E] at ../libs/gst/base/gstbasesrc.c:3127 - "<udpsrc1> error: Internal data stream error."
[E] at ../libs/gst/base/gstbasesrc.c:3127 - "<udpsrc1> error: streaming stopped, reason not-negotiated (-4)"
[!] at C:\projects\qgroundcontrol\src\VideoReceiver\GstVideoReceiver.cc:1305 - "GStreamer error: Internal data stream error."
[E] at ../gst/rtsp/gstrtspsrc.c:6469 - "<source> server closed connection"
[E] at ../gst/rtsp/gstrtspsrc.c:6469 - "<source> server closed connection"
[E] at ../gst/rtsp/gstrtspsrc.c:6506 - "<source> error: Could not receive message. (Received end-of-file)"
[E] at ../gst/rtsp/gstrtspsrc.c:8215 - "<source> error: Could not send message. (Received end-of-file)"
```

LorenzMeier, 2021-02-16:
We're focusing video support on RTSP / H264/H265 going forward. Anything else is experimental.

closed 2021-02-16 by LorenzMeier