mavlink/qgroundcontrol #9418: Unable to get a bigger attitude and heading indicator

dorovl, 2021-02-06:
Prior to the version v4.1.0, to get a bigger attitude and heading indicator, just check the "Use Vertical Instrument Panel". Now if "Use Vertical Instrument Panel" is checked, the attitude and heading indicator disappears... :cry:

Note: the attitude and heading indicator is not really a "vertical instrument" like a vertical speed or glide path indicator. Therefore I suggest to rename this indicator to "attitude and heading indicator".

Antiheavy, 2021-02-07:
I just noticed this today as well while testing the newest stable release.  I know that some people really like the larger display of the Vertical Instrument Panel:
[image]
[image]

Antiheavy, 2021-02-07:
> Note: the attitude and heading indicator is not really a "vertical instrument" like a vertical speed or glide path indicator. Therefore I suggest to rename this indicator to "attitude and heading indicator".

or "Larger Instrument Panel"

dorovl, 2021-02-08:
> or "Larger Instrument Panel"

Yes, just perfect!:+1:

DonLakeFlyer, 2021-02-18:
Dup of #9405

closed 2021-02-18 by DonLakeFlyer